PX4/PX4-Autopilot #8320: Multi vehicle can not be added

PujieXin, 2017-11-19:
Wrong:
expected exactly one input file as argument
while processing /home/ise/src/Firmware/launch/single_vehcile_spawn.launch:
Invalid <param> tag: Cannot load command parameter [iris_1_sdf]: command [/opt/ros/kinetic/share/xacro/xacro.py /home/ise/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro rotors_description_dir:=/home/ise/src/Firmware/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=14560 > iris_1.urdf ; 'gz sdf -p iris_1.urdf'] returned with code [2]. 

Param xml is <param command="$(arg cmd)" name="$(arg vehicle)_$(arg ID)_sdf"/>
The traceback for the exception was written to the log file.

[roslaunch][ERROR] 2017-11-19 17:46:03,929: while processing /home/ise/src/Firmware/launch/single_vehcile_spawn.launch:
Invalid <param> tag: Cannot load command parameter [iris_1_sdf]: command [/opt/ros/kinetic/share/xacro/xacro.py /home/ise/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro rotors_description_dir:=/home/ise/src/Firmware/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=14560 > iris_1.urdf ; 'gz sdf -p iris_1.urdf'] returned with code [2]. 

Param xml is <param command="$(arg cmd)" name="$(arg vehicle)_$(arg ID)_sdf"/>
[roslaunch][ERROR] 2017-11-19 17:46:03,929: The traceback for the exception was written to the log file
[roslaunch][ERROR] 2017-11-19 17:46:03,929: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 307, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 453, in load_config_default
    raise RLException(e)
RLException: while processing /home/ise/src/Firmware/launch/single_vehcile_spawn.launch:
Invalid <param> tag: Cannot load command parameter [iris_1_sdf]: command [/opt/ros/kinetic/share/xacro/xacro.py /home/ise/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro rotors_description_dir:=/home/ise/src/Firmware/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=14560 > iris_1.urdf ; 'gz sdf -p iris_1.urdf'] returned with code [2]. 

Param xml is <param command="$(arg cmd)" name="$(arg vehicle)_$(arg ID)_sdf"/>

[rospy.core][INFO] 2017-11-19 17:46:03,929: signal_shutdown [atexit]

I can launch single vehicle.There is no problem.
Do you know what's the problem.Thank you.

Seiwert, 2017-12-12:
This is not an ideal solution and I'm not endorsing it by any means but this should get you going.

1. Create a bash script that runs xacro. Here is my `workaround.sh` (put it in /src/Firmware/): 
``` sh
# $1 = find px4
# $2 = vehicle
# $3 = mavlink_udp_port
# $4 = ID
source /opt/ros/kinetic/setup.bash
rosrun xacro xacro $1/Tools/sitl_gazebo/models/rotors_description/urdf/$2_base.xacro rotors_description_dir:=$1/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=$3 > $2_$4.urdf
```

2. Then change the `single_vehcile_spawn.launch` to be this:
```xml
<launch>

    <!-- Posix SITL environment launch script -->
    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>
    <arg name="est" default="lpe"/>
    <arg name="vehicle" default="iris"/>
    <arg name="ID" default="1"/>
    <arg name="rcS" default="$(find px4)/posix-configs/SITL/init/$(arg est)/$(arg vehicle)_$(arg ID)"/>
    <arg name="mavlink_udp_port" default="14560" />

    
    <param name="launchWorkAround" command="rosrun px4 workaround.sh $(find px4) $(arg vehicle) $(arg mavlink_udp_port) $(arg ID)" />

    <arg name="cmd" default="gz sdf -p $(arg vehicle)_$(arg ID).urdf" />

    <param command="$(arg cmd)" name="$(arg vehicle)_$(arg ID)_sdf" />

    <node name="sitl_$(arg ID)" pkg="px4" type="px4" output="screen"
        args="$(find px4) $(arg rcS)">
    </node>

    <node name="$(arg vehicle)_$(arg ID)_spawn" output="screen" pkg="gazebo_ros" type="spawn_model"
        args="-sdf -param $(arg vehicle)_$(arg ID)_sdf -model $(arg vehicle)_$(arg ID) -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)" respawn="false"/>

</launch>

<!-- vim: set et ft=xml fenc=utf-8 ff=unix sts=0 sw=4 ts=4 : -->
```

This is definitely **not** the right answer but it is something I suppose...

Cheers

lamping7, 2018-03-16:
This was fixed. https://github.com/PX4/Firmware/pull/8831

closed 2018-03-16 by lamping7

weiminshen99, 2018-03-20:
Dear Anthony,
Thanks for fixing this. I am new for this and have a naive question: How do I download your latest fix from https://github.com/px4/firmware? If I do a "git clone" there, would it include your fix (#8831)? Thanks!
Wingman

TobiMiller, 2018-03-21:
@weiminshen99 
on the top you see the badge with label "merged". this means that the Pull Request is merged into the master. so it is already in the master repository