PX4/PX4-Autopilot #8329: Simulate in Gazebo with ideal condition

TokyoClod, 2017-11-21:
When I used the offboard mode in Gazebo to simulate the px4, i found it would oscillate in a small range after specified a certain pose (as the example in tutorial [0,0,2]). For the use of testing the algorithm of visual landing, i hope that the px4 can follow the command exactly.  
I find the config file in [Firmware/posix-configs/SITL/init/lpe/iris](url) and [mavros/mavros/launch/px4_config.yaml](url) may be associated with this,but i don't know how to modify it.
Could any one show me how to achieve it? Hoping for any solution!

TSC21, 2017-11-21:
@TokyoClod your question is not very clear. You want to know what do you have to modify/tweak to have the vehicle follow the command "exactly"? What does exactly mean?

TokyoClod, 2017-11-21:
@TSC21 I want the px4 could stay static at the the setpoint without oscillation shown as following gif
[image]
I used position controller in offboard mode, setting the pose = [0,0,2].

TSC21, 2017-11-21:
It won't be static cause we have noise simulated on the sensors side.

TokyoClod, 2017-11-21:
Then can i set the noise of sensors to zero?
I guess that it's in the config file of [https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/lpe/iris](url). But i can't understand it clearly so don't know how to modify it.

ndepal, 2017-11-22:
GPS noise is added in the `gazebo_mavlink_interface`: https://github.com/PX4/sitl_gazebo/blob/master/include/gazebo_mavlink_interface.h#L173

You can try setting both `gps_random_walk` and `gps_noise_density` to 0.

TokyoClod, 2017-11-23:
@ndepal 
I cloned the latest version of Firmware and initialize the sitl_gazebo by 
`git submodule update --init --recursive`
It happens that my `gazebo_mavlink_interface.h` just  has `ev_random_walk ` and `ev_noise_density`, has no  `gps_random_walk` and `gps_noise_density`?  I don't know why....
By setting `ev_random_walk ` and `ev_noise_density` to zero, it doesn't make any change in Gazebo simulation. Do you have any ideas?

However, by comment [https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_imu_plugin.cpp#L286](url), it behaves much better  in herizon, by still shave in vertical direction. As following:
[image]

I think the noise may come from the Gazebo simulator? Keep trying to cancel the noise now...

ndepal, 2017-11-24:
Ok, so the gps noise which has been added to SITL gazebo recently is not actually used yet in Firmware master, so you are getting noise-free GPS measurements from the simulator.

closed 2018-02-24 by PX4BuildBot