PX4/PX4-Autopilot #8330: SITL compilation bug

hiro64, 2017-11-21:
I just cloned the newest version and I am trying to compile the SITL simulation with Gazebo by running `make posix_sitl_default gazebo`. However, I am receiving some errors related to the mavlink libraries. It seems the mavlink_msg_hil_actuator_controls.h file from the mavlink folder is not being included. One of the errors when running make is below:

> Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:903:8: error:
> MAVLINK_MSG_ID_HIL_ACTUATOR_CONTROLS’ was not declared in this scope
> case MAVLINK_MSG_ID_HIL_ACTUATOR_CONTROLS:

I tried it with last week's master and got the same error, but I compiled it with no problems back in September, so I assume something must have happened in the meantime.

TSC21, 2017-11-21:
Did you update all the submodules accordingly?

hiro64, 2017-11-21:
Yes, I ran `git submodule update --init --recursive` and they were updated with no problems.

TokyoClod, 2017-11-22:
i find the latest version Firmware missing the iris.sdf in [https://github.com/PX4/sitl_gazebo/tree/b052c97f7c3a2c39ab8ec06ae79c66431f16c659/models/iris](url)

TSC21, 2017-11-24:
@hiro64 it looks like you are don't have the Mavlink submodule updated. Try `make distclean`, then `git submodule sync --recursive` and then `git submodule update --init --recursive`.

hiro64, 2017-11-28:
@TSC21 I cloned it again into a new folder and ran these commands but it ended up going through the same problem... could it be something in my machine?

closed 2018-08-06 by santiago3dr