PX4/PX4-Autopilot #8333: [SubmodulesClean] error 1

5t0ic, 2017-11-22:
The initial setup is not working whenever I try to "make distclean" Its consistently indicating error 1 when executed in the px4 console. When the command is issued from the custom eclipse ide it gives me a c/c++ error and sates "make *** [submodulesclean] Error 1". 

The "make clean" function works but it doesn't help with the issue.

dagar, 2017-11-22:
Which platform?

Does this work? `git submodule update --init --recursive`

dagar, 2017-11-26:
Can you provide any update? Have you gone through the full toolchain installation? https://dev.px4.io/en/setup/dev_env.html

closed 2017-12-06 by dagar

Dream123456, 2018-01-09:
I also got into the same trouble. How to do?

dagar, 2018-01-09:
Which platform? Can you try `make distclean`?

Dream123456, 2018-01-11:
will, I till try use it in win10, but I try Ubuntu today, I will update later. Thx.

Dream123456, 2018-01-11:
win10, I will try use it in win10, but I using Ubuntu 16 today, I will update later. Thx.

aogrcs, 2018-01-29:
I used win10, when tried make distclean, it said as following:
make distclean 
Removing uavcan/__init__.pyc
Removing uavcan/dsdl/__init__.pyc
Removing uavcan/dsdl/common.pyc
Removing uavcan/dsdl/parser.pyc
Removing uavcan/dsdl/signature.pyc
Removing uavcan/dsdl/type_limits.pyc
Removing uavcan/transport.pyc
Usage: git submodule [--quiet] add [-b branch] [-f|--force] [--reference <repository>] [--] <repository> [<path>]
   or: git submodule [--quiet] status [--cached] [--recursive] [--] [<path>...]
   or: git submodule [--quiet] init [--] [<path>...]
   or: git submodule [--quiet] update [--init] [-N|--no-fetch] [-f|--force] [--rebase] [--reference <repository>] [--merge] [--recursive] [--] [<path>...]
   or: git submodule [--quiet] summary [--cached|--files] [--summary-limit <n>] [commit] [--] [<path>...]
   or: git submodule [--quiet] foreach [--recursive] <command>
   or: git submodule [--quiet] sync [--] [<path>...]
make: *** [submodulesclean] Error 1

nanthony21, 2018-01-31:
I'm getting this same issue on Windows 7. I've update the submodules as recommended and git shows that everything is up to date.  The error doesn't occur if I try it on recent commits to master . But if I try to run it on the commit that is tagged "1.7.3" or anything earlier I get the error.

Note: Even though distclean succeeds on recent commits, the build still fails with "Error 2: you must run this command from the toplevel of the working tree."

dagar, 2018-01-31:
If you can post full output and likely add some debugging as needed we can probably work through it. Some of this sounds like old versions of git not liking certain arguments, other issues could be with windows paths.

dagar, 2018-01-31:
Please open a fresh issue and tag me if you'd like to continue debugging.

gadavidd, 2018-03-27:
Hey, please help me! I have same problems on Windows 8, Could you help me?