PX4/PX4-Autopilot #8349: Unsupported FCU Error MAVROS

KochC, 2017-11-24:
**Bug Report**
The ```roslaunch mavros px4.launch``` with the adaption for my serial device ttyS0 works well.
I get the following outputs after running this:
```
~/ros_ws$ roslaunch mavros px4.launch 
... logging to /home/ubuntu/.ros/log/9bd7afb2-d0fc-11e7-91d4-b827eb6b9362/roslaunch-ubuntu-2053.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:55070/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_url: /dev/ttyS0:57600
 * /mavros/gcs_url: 
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
[... 454 lines cut ...]
[ INFO] [1511517236.647842813]: Plugin waypoint initialized
[ INFO] [1511517236.648183744]: Autostarting mavlink via USB on PX4
[ INFO] [1511517236.648478061]: Built-in SIMD instructions: None
[ INFO] [1511517236.648646391]: Built-in MAVLink package version: 2017.10.10
[ INFO] [1511517236.648859408]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1511517236.649032738]: MAVROS started. MY ID 1.240, TARGET ID 1.1
[ERROR] [1511517243.676111562]: MODE: Unsupported FCU
```

```
header: 
  seq: 0
  stamp: 
    secs: 1511517560
    nsecs: 532773555
  frame_id: ''
status: 
  - 
    level: 0
    name: mavros: FCU connection
    message: Node starting up
    hardware_id: ''
    values: []
---
```

TSC21, 2017-11-24:
What's your flight controller?

vooon, 2017-11-24:
You do not have a connection with FCU. After `MAVROS started` should be `CON: Got HEARTBEAT`.
Also your offboard node should subscribe to `/mavros/state` and wait [State.connected == true][1].

[1]: http://docs.ros.org/kinetic/api/mavros_msgs/html/msg/State.html

KochC, 2017-11-24:
ok so that means even the ```px4.launch```dose not work properly with my configuration.
I'm using a DroPix with the default PX4 FMU4 Firmware.

KochC, 2017-11-27:
I connected the Telemetry2 from the Pixhawk to the Raspberry Pi
```
PX RX <-> Pi TX
PX TX <-> Pi RX
GND <-> GND
VCC <-> VCC
```

On my raspberry pi I can use ```ttyS0``` or ```serial0``` for the serial device. It is also switched on and its possible to read from. My user got the rights to read and write.

My Pixhawk is set to 912600 Baud 8N1 and my Raspberry is the same.

What else do I have to do?

vooon, 2017-11-27:
Do not connect VCC - or power source of Pi or PH may blow.
Other pin correct.

Do you setup Pi3 to disable bluetooth and remap pl011 serial to header pins?
Also Pi2 need special lines in config.txt to be able run at 921600, not sure about Pi3.

KochC, 2017-11-28:
@vooon ok so because I'm not sure about the configuration of the ttyS0 port I tried to use a FTDI via USB.
```
Pi -> USB -> FTDI -> GND / RX & TX -> DroPix
```
I powered the Dropix by a separate USB and the Pi also with a separate USB.

px4.launch output is 
```
~/ros_ws$ roslaunch my package px4.launch 
... logging to /home/ubuntu/.ros/log/7b70b9d0-d435-11e7-8fdf-b827eb6b9362/roslaunch-mypackage-5183.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:39434/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: ROLL_180
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: ROLL_180
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: ROLL_180
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_url: /dev/ttyUSB0:921600
 * /mavros/gcs_url: udp://:14556@192....
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/rot_covariance: 99999.0
[... 180 lines cut ...]
[ INFO] [1511871190.192506898]: Plugin setpoint_velocity initialized
[ INFO] [1511871190.193853760]: Plugin sys_status loaded
[ INFO] [1511871190.265217187]: Plugin sys_status initialized
[ INFO] [1511871190.266466029]: Plugin sys_time loaded
[ INFO] [1511871190.307360530]: TM: Timesync mode: MAVLINK
[ INFO] [1511871190.315122695]: Plugin sys_time initialized
[ INFO] [1511871190.316359974]: Plugin vfr_hud loaded
[ INFO] [1511871190.331181134]: Plugin vfr_hud initialized
[ INFO] [1511871190.331470454]: Plugin vibration blacklisted
[ INFO] [1511871190.332400653]: Plugin vision_pose_estimate loaded
[ INFO] [1511871190.438018697]: Plugin vision_pose_estimate initialized
[ INFO] [1511871190.439162384]: Plugin vision_speed_estimate loaded
[ INFO] [1511871190.482894722]: Plugin vision_speed_estimate initialized
[ INFO] [1511871190.484180751]: Plugin waypoint loaded
[ INFO] [1511871190.535862284]: Plugin waypoint initialized
[ INFO] [1511871190.536223166]: Autostarting mavlink via USB on PX4
[ INFO] [1511871190.536555767]: Built-in SIMD instructions: None
[ INFO] [1511871190.536766234]: Built-in MAVLink package version: 2017.10.10
[ INFO] [1511871190.536977534]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1511871190.537179146]: MAVROS started. MY ID 1.240, TARGET ID 1.50
```

so again no Heartbeat received.

Is it possible that its a PX4 problem?

TSC21, 2017-11-28:
Do you have latest upstream master px4fmuv2_default loaded into the Dropix?

KochC, 2017-11-28:
@TSC21 yes

TSC21, 2017-11-28:
What's the value of your `SYS_COMPANION` parameter currently?

KochC, 2017-11-28:
@TSC21 ISYS_COMPANION is ```Companion Link (921600 baud, 8N1)```

KochC, 2017-11-28:
well I again flashed the firmware, calibrated all sensors new, set the ```SYS_COMPANION``` to ```57600``` baud and adapted my launch file also to ```57600```baud and its working fine now 😂
@TSC21 and @voon, thanks a lot for your help.

```
[ INFO] [1511876698.657583211]: Known MAVLink dialects: common ardupilotmega ASLUAV autoquad matrixpilot paparazzi slugs standard uAvionix ualberta
[ INFO] [1511876698.657855024]: MAVROS started. MY ID 1.240, TARGET ID 1.50
[ WARN] [1511876698.674926013]: TM: Clock skew detected (-1511876484.446492434 s). Hard syncing clocks.
[ INFO] [1511876699.630708587]: CON: Got HEARTBEAT, connected. FCU: PX4 Autopilot
[ INFO] [1511876699.790941468]: IMU: High resolution IMU detected!
[ INFO] [1511876700.680052534]: VER: 1.1: Capabilities         0x000000000000e4ef
[ INFO] [1511876700.680381324]: VER: 1.1: Flight software:     010700c0 (0000000076F2E0A7)
[ INFO] [1511876700.680577463]: VER: 1.1: Middleware software: 010700c0 (0000000000000001)
[ INFO] [1511876700.681345769]: VER: 1.1: OS software:         071600ff (0000000000000001)
[ INFO] [1511876700.681668101]: VER: 1.1: Board hardware:      00000011
[ INFO] [1511876700.681895229]: VER: 1.1: VID/PID:             26ac:0011
[ INFO] [1511876700.682076681]: VER: 1.1: UID:                 3334510b32363238
```

closed 2017-11-28 by KochC