PX4/PX4-Autopilot #8358: The new STM32H743VIT6 can be included as the variant px4fmu-v5-pro

zarathustr, 2017-11-25:
ST has released its most powerful processor STM32H7. The core clock speed can be up to 400MHz plus 2MB flash and 1MB ram. According to ST，the programming architecture of H7 will exactly be the same with F7. That is to say the ISR configuration and PIN would be consistent. So a suggestion is that some tiny revisions be added based on the px4fmu-v5，which supports the clock speed and ram size. Then just test H7 to see if it is OK for programming with existing NuttX RTOS for stm32f7.

LorenzMeier, 2017-11-25:
We have already dev boards and will bring them up starting next week.

yangang123, 2017-12-01:
good idea

zarathustr, 2017-12-01:
@LorenzMeier  Hi Lorenz, we have started the design of STM32H7 autopilot. We will mail you one once finished

zarathustr, 2017-12-23:
@LorenzMeier  Hi Lorenz, Nice to see you. We have finished the H7 autopilot board. Should I mail you one for test?

LorenzMeier, 2017-12-23:
@zarathustr Could you drop me an email on lorenz@px4.io? I can then send you the shipping address.

closed 2018-08-06 by santiago3dr

bys1123, 2019-01-25:
@zarathustr I like your idea, did you already try porting to STM32H7? Are you success?

zarathustr, 2019-01-25:
@bys1123 The current mainstream of nuttx on STM32H7 is still undergoing initial experimentation. Many modules including SDMMC, QSPI, FDCAN have not been tested yet. Also, there are many bugs insides the official STM32CubeMx Lib which may set up obstacles for those contributors to push forward.