PX4/PX4-Autopilot #8361: Fixed Wing reject auto mission

jinchengde, 2017-11-26:
fixed wing cannot entry into auto mission

here is the flight log
https://logs.px4.io/plot_app?log=90d6dcd2-7b87-4963-9390-987c1d062b4a
https://logs.px4.io/plot_app?log=27b7223c-7d7b-4dba-b82a-da3229e546e0

dagar, 2017-11-26:
Auto mission is being rejected because you don't have a valid position estimate.

[image]

[image]

[6fb19372-e027-4b23-b0e8-5425d409f5f3.ulg.pdf](https://github.com/PX4/Firmware/files/1504138/6fb19372-e027-4b23-b0e8-5425d409f5f3.ulg.pdf)

@priseborough could you take a look at these logs?

priseborough, 2017-11-26:
The inertial navigation is failing due to high vibration You need to reduce vibration levels by correct isolation of the autopilot.

Very high vibration levels
[image]

GPS vertical velocity and baro vertical position innovations are diverging in the same direction
[image]

closed 2017-11-26 by priseborough

jinchengde, 2017-11-27:
thanks @priseborough