PX4/PX4-Autopilot #8367: Potential critical NuttX alignment issue

LorenzMeier, 2017-11-26:
The full report is here and I'm including it to ensure that we're not forgetting about it:
https://bitbucket.org/nuttx/nuttx/issues/78/possible-bss-override-by-stack-coloration

@davids5 @dagar I would appreciate your review.

Thanks!

matanhavi, 2017-11-26:
Note that for what I seen so far, for px4fmu there should be no issue, because the last var on the bss is _io_timer_channel_stats_, which is 4 byte aligned.
I have seen the issue because I didn't set BOARD_HAS_CAPTURE, '_once_' var has become the last one, and its uint8_t

BazookaJoe1900, 2017-11-26:
Note that for what I seen so far, for px4fmu there should be no issue, because the last var on the bss is io_timer_channel_stats, which is 4 byte aligned.
I have seen the issue because I didn't set BOARD_HAS_CAPTURE, 'once' var has become the last one, and its uint8_t

dagar, 2017-11-26:
Add ALIGN to the linker scripts? https://github.com/PX4/Firmware/blob/master/nuttx-configs/px4fmu-v2/scripts/ld.script#L143

BazookaJoe1900, 2017-11-26:
Yes, I think that adding
. = ALIGN(4);
Should to the work.
Can you confirm?

davids5, 2017-11-26:
I will check and confirm it in the AM tomorrow.

LorenzMeier, 2017-11-27:
@davids5 Did you have a chance to have a look?

davids5, 2017-11-27:
@LorenzMeier yes. Current build is not affected, but I will do a PR to avoid any future issues.

LorenzMeier, 2017-11-27:
Moving to v1.8.

dagar, 2017-12-06:
Fixed in https://github.com/PX4/Firmware/pull/8382.

closed 2017-12-06 by dagar

matanhavi, 2017-12-07:
@davids5 Thanks!

BazookaJoe1900, 2017-12-07:
@davids5 Thanks!