PX4/PX4-Autopilot #8368: Multi-vehicle(Swarm UAV) Simulation Error

pineland, 2017-11-26:
Hi,
I'm trying to simulate multi-vehicle simulation under the following environment:
- Ubuntu 16.04 LTS, ROS kinetic, Gazebo 7
- guided on [the multi-vehicles simulation of Dronecode](https://dev.px4.io/en/simulation/multi-vehicle-simulation.html).

As guided by the procedure, I was faced with the following problem which occurred when I run 'roslaunch px4 multi_uav_mavros_sitl.launch'.

The error was the following:

Usage: xacro.py [options] <input>

xacro.py: error: expected exactly one input file as argument
while processing /home/burobotics/Guang/src/Firmware/launch/single_vehcile_spawn.launch:
Invalid <param> tag: Cannot load command parameter [iris_1_sdf]: command [/opt/ros/kinetic/share/xacro/xacro.py /home/burobotics/Guang/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro rotors_description_dir:=/home/burobotics/Guang/src/Firmware/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=14560 > iris_1.urdf ; 'gz sdf -p iris_1.urdf'] returned with code [2]. 

Then, the error was solved by modifying ~/src/Firmware/launch/single_vehcile_spawn.launch as follow :

<arg name="cmd" default="$(find xacro)/xacro.py $(find px4)/Tools/sitl_gazebo/models/rotors_description/urdf/$(arg vehicle)_base.xacro rotors_description_dir:=$(find px4)/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=$(arg mavlink_udp_port) > $(arg vehicle)_$(arg ID).urdf ; 'gz sdf -p $(arg vehicle)_$(arg ID).urdf'" />
->
<arg name="cmd" default="$(find xacro)/xacro.py $(find px4)/Tools/sitl_gazebo/models/rotors_description/urdf/$(arg vehicle)_base.xacro rotors_description_dir:=$(find px4)/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=$(arg mavlink_udp_port)" />

Now I can see that multi vehicles appear in the gazebo. However, I am having trouble getting them to move.

In order to work them by a mavros node, I got a [pub_setpoints_pos.cpp](https://github.com/Jaeyoung-Lim/modudculab_ros) which make multi-vehicles hovering to a height of 2 meters, changed CMakeLists.txt, successfully built, and changed the node.launch as follow :

===============<~/catkin_ws/src/mavros/mavros/CMakeLists.txt>================
add_executable(pub_setpoints_pos src/pub_setpoints_pos.cpp)
target_link_libraries(pub_setpoints_pos ${catkin_LIBRARIES})
add_dependencies(pub_setpoints_pos mavros)

==============<~/catkin_ws/src/mavros/mavros/launch/node.launch>==============

<launch>
	
	

	<arg name="fcu_url" />
	<arg name="gcs_url" />
	<arg name="tgt_system" />
	<arg name="tgt_component" />
	<arg name="pluginlists_yaml" />
	<arg name="config_yaml" />
	<arg name="log_output" default="screen" />

	<node pkg="mavros" type="mavros_node" name="mavros" required="true" clear_params="true" output="$(arg log_output)">
		<param name="fcu_url" value="$(arg fcu_url)" />
		<param name="gcs_url" value="$(arg gcs_url)" />
		<param name="target_system_id" value="$(arg tgt_system)" />
		<param name="target_component_id" value="$(arg tgt_component)" />

		
		<rosparam command="load" file="$(arg pluginlists_yaml)" />
		<rosparam command="load" file="$(arg config_yaml)" />
	</node>

	
	<node name="pub_setpoints" pkg="mavros" type="pub_setpoints_pos" output="screen">

	</node>
	

</launch>

To test, the following commands finally are executed and the multi-vehicles are not working.

- 1 terminal : roscore
- 2 terminal : 
cd ~/src/Firmware
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 multi_uav_mavros_sitl.launch   <- 2 iris quadcopters appear
- 3 terminal : 
rosrun mavros mavsafety arm      **<- service [/mavros/cmd/arming] unavailable** 
rosrun mavros mavsys mode -c OFFBOARD'    **<- service [/mavros/cmd/arming] unavailable** 

How can I get those vehicles to work? Do you have any comments or examples?
I really appreciate your help in advance.

closed 2017-12-10 by pineland

CrazyDominic, 2018-01-12:
Is it solved? I have the same problem. If you solved it, plz give me some advice.

billiepratama, 2019-01-28:
I got the same problem too. The Gazebo works well and multi vehicles appear, but mavros doesn't get the connection. Please help me. :(

hamishwillee, 2019-01-28:
@billiepratama This is a couple of years old and was closed at the time. Whatever issue you have is probably unrelated, but even if it is, might be on different environment. Please create a new issue with all the detail about your particular setup.

billiepratama, 2019-02-13:
> I am facing the same issue, 3 drones are launched in gazebo but doesn't
> get connected to QGC

Do you follow this link tutorial?
https://dev.px4.io/en/simulation/multi-vehicle-simulation.html

I've tried to change the UDP port in the rcS file, but it's still failed.
I've already searched everywhere, but I don't find any good solution