PX4/PX4-Autopilot #8380: FW crash v1.0.7-rc2

r0gelion, 2017-11-28:
Phantom flying wing with Pixhawk mini autopilot running v1.7.0-rc2 (V3).

After 6 successful flights, during the 7th flight, the pilot took off in manual mode, switched to Auto and the plane accelerated to maximum thrust uncontrollably; once switched back into manual the pilot reported only partial control, attempted to land manually but crashed due to the vast acceleration and only partial control over the craft.

[Log](https://review.px4.io/plot_app?log=6fb19372-e027-4b23-b0e8-5425d409f5f3)

[Video](https://youtu.be/3qIzzh4hfUs)

@dagar

dagar, 2017-11-28:
I saw this log a couple days ago. The major thing that stands out is that the pitch demand is being ignored. There's also a sizable roll offset.

[image]
 
The other minor thing is that the maximum roll max is set to 0 (unconstrained). Resetting to the default should be fine.

[image]

How badly damaged was the vehicle? I suspect some kind of mechanical failure. Servos, linkages, control surfaces, or possibly even the battery coming loose and moving around.

bresch, 2017-12-04:
@r0gelion Link of the log is broken. Could you update it please?

bkueng, 2017-12-04:
@bresch Works again. Sorry I messed up an upgrade.

bresch, 2017-12-04:
@bkueng Thanks!

acfloria, 2017-12-04:
@r0gelion It is probably not related to the crash but which airspeed sensor are you using?
I am asking because this value is quite high: `SENS_DPRES_OFF, 122.68370819091797`

r0gelion, 2017-12-04:
@acfloria this is the airspeed used.

[image]

Thanks!

dagar, 2017-12-04:
@acfloria in my experience that's a pretty typical offset for an MS4525.

dagar, 2017-12-08:
Most likely mechanical failure.

closed 2017-12-08 by dagar