PX4/PX4-Autopilot #8392: F450 with Pixhawk 3 Pro v1.7.0-rc3 crash

r0gelion, 2017-11-29:
Flight of DJI F450 with Pixhawk 3 Pro autopilot using v1.7.0-rc3 (66d4a1b3)

The pilot reported the vehicle took off, switched to Auto, reached the first waypoint and it started to climb above 300 meters, the pilot commanded RTL through QGC, the vehicle started to move towards home position but kept gaining altitude. The pilot sent land command through QGC but the vehicle didn't respond, it only responded to RC roll so the pilot engaged the kill switch.

[Log](https://review.px4.io/plot_app?log=0cdfec08-8848-4892-bcff-d4a720e99803)

@dagar

dagar, 2017-11-30:
What history does this vehicle have? Does it have previous successful flights?

In mission mode when it was stuck in the climb the commanded throttle (blue line) went to the minimum (12%).

[image]

Combined with the relatively high PWM_MIN of 1230 us the minimum motor outputs were around 1300us.

[image]

Which ESCs does this vehicle have? Do they need to be calibrated? I would start with a short manual (stabilized) checkout flight.

r0gelion, 2017-11-30:
That vehicle had previous successful flights with v1.7.0-rc1 (3744cac)
https://review.px4.io/plot_app?log=be5c0de0-606e-4279-b196-327e76d08d13
https://review.px4.io/plot_app?log=c8b76678-ccba-424b-9d7a-83e87428e8e2
https://review.px4.io/plot_app?log=ac3eb2a8-e718-4a24-9044-a64320994d67
https://review.px4.io/plot_app?log=29d7b3ee-3a91-469c-910f-6e3a1b77ba43
https://review.px4.io/plot_app?log=344bef4b-1f2e-4047-8503-6e16028db80b
https://review.px4.io/plot_app?log=7fbe2e3e-60b9-4678-b760-ca7511745e1d
https://review.px4.io/plot_app?log=b69c8474-9cb4-44e5-bc4d-de8be8c9f968
https://review.px4.io/plot_app?log=38a302bd-0cbd-4cc4-a885-f2036a086304
https://review.px4.io/plot_app?log=0ccc1b69-8b09-430f-a994-7ea2c9cb5091
https://review.px4.io/plot_app?log=c9686972-8334-4c33-ad8d-38b612798308

These are the ESCs used on this vehicle. All calibrations performed.
[image]

xdwgood, 2017-12-01:
@dagar  Is the green line a set height for the current task?  thanks :smiley:

LorenzMeier, 2017-12-02:
@MaEtUgR @Stifael Could you please have a look?

dagar, 2017-12-08:
Let's go over this again in detail tomorrow, but so far it looks like a physical setup issue. In the previous flight logs the nominal pwm output in a climb, hover, or forward flight is higher than the "flyaway" value in this log.

LorenzMeier, 2017-12-08:
@Stifael Could you take another pass?

Stifael, 2017-12-08:
There is not much I can add to what @dagar  already mentioned.
I looked at the entire pipeline:
for auto:
position setpoint -> velocity setpoint -> thrust setpoint -> thrust magnitude from vehicle_rates_septoint_s -> actuator controls -> actuator outputs
for manual:
stick inputs -> velocity setpoints -> ... (as above)

All the above messages seem to be correct in terms of flight intention and control outputs (they all go basically to the minimum). 
@r0gelion do you know why the pwm minimum is set to 1230? Was it set that high on purpose? Is there anything that changed in between the crash and the previous flights such as weight? In the log files from previous flights the average pwm output is around 1500, but in the crash log the pwm output is in average at around 1325.

bresch, 2017-12-08:
@Stifael The PWM min is set to 1230 because it's the minimal value of the DJI ESCs and they don't support calibration.
If fact, that could be the issue: the ESCs are not the DJI ones. With this value, the lifting system produces more thrust than the weight of the drone.

LorenzMeier, 2017-12-10:
Closing as misconfigured.

closed 2017-12-10 by LorenzMeier