PX4/PX4-Autopilot #8398: F450 with Pixhawk 1 - f63c821 crash

r0gelion, 2017-12-01:
The pilot reported it was flying in Auto and was seing the vehicle following the mission in QGC, then the RC signal was lost and the vehicle crashed on the ceiling of a building.

This a screen capture of the messages in QGC
[image]

[Log](https://review.px4.io/plot_app?log=c087212c-24ec-4076-b83c-0c690b5c26ba)

Previous successful fligths the same day
https://review.px4.io/plot_app?log=a768ef29-6be8-4245-b191-183ffa20ddb6
https://review.px4.io/plot_app?log=56d2c68b-cff0-4198-bb33-4197e44a4815
https://review.px4.io/plot_app?log=95e5f4c8-b294-4182-b630-256600424117
https://review.px4.io/plot_app?log=a75645fa-3af5-494a-b3cf-748f5d9a6d89
https://review.px4.io/plot_app?log=c7a0d5c4-f79a-47d2-83f6-0a8d94fd216c
https://review.px4.io/plot_app?log=83b00da7-f696-4af4-b3b7-f768fead3c7f

@dagar

xdwgood, 2017-12-01:
Hi @r0gelion , I have some analysis, I hope I can help you  :smiley: 
[image]
I downloaded the log and got the image above 
I think the drone was normal throughout the flight . The pilot takes off in mission mode and the drone performs the map task well . The pilot switches mode to position mode at 08:36.500 and stops the mission mode automatic flight 
Look this ： 
[image]
The drone then flies in position mode, according to the pilot's rc input .Pilot switch mode to altitiude mode at 09:50 .At 09:53, the drone strikes the building (under the pilot)  and automatically executes the RTL(because rc lost). Crash into the building and crash while performing RTL 
Look this ：
[image]

**Summed up ：**
Before the drone strikes the building at 09:53, the drone behaves normally (good Angle, location, speed tracking). The Rc was damaged after the collision, so the drone attempted to perform RTL. 
But the impact damaged the drone's power output, causing the drone to crash on the ground. **So I personally think this is a pilot mistake.**  :smiley:

r0gelion, 2017-12-01:
@xdwgood Thank you for the analysis, I checked with the pilot and it makes sense.

I'm closing this issue.

closed 2017-12-01 by r0gelion