PX4/PX4-Autopilot #8400: MR: uncontrollable altitude gain when engaging auto or rtl

santiago3dr, 2017-12-01:
Sequence of events

armed in position mode
engaged auto, mission rejected
took off in position mode
second attempt at engaging auto
copter took off with uncontrollable gain in altitude

rtl presented the same behavior
https://logs.px4.io/plot_app?log=01fbca4e-353b-4874-831a-03f091098584

@dagar

santiago3dr, 2017-12-01:
similar events on the snapdragon flight controller (eagle)
https://logs.px4.io/plot_app?log=2943d850-0853-40f3-b129-63f6e4c1e29e

santiago3dr, 2017-12-01:
was able to replicate the issue by engaging Auto once prior to obtaining global position and a second time once gps lock has been achieved:
https://logs.px4.io/plot_app?log=06bbc8b5-bf7d-4852-9001-8bffdc641596

LorenzMeier, 2017-12-02:
@Stifael @MaEtUgR Could you please have a look?

Stifael, 2017-12-04:
this behavior is reproducible in sitl: as @santiago3dr pointed out, if the pilot takes off in manual mode, the gps lock will occur in air with home.alt set to some large number. The pilot can now switch to auto mode (for instance hold) and the vehicle will gain in altitude. 

@priseborough what is the desired behavior? Is it intended to have a home position set when vehicle is already in air? 
The easiest solution for now would be to not set home position when already in air. This way the pilot cannot switch to auto mode if he took of in manual mode but no gps lock

Stifael, 2017-12-04:
could be related #8392...

LorenzMeier, 2017-12-04:
The altitude should be reset. Taking off without GPS but switching to auto in air has to be supported.

LorenzMeier, 2017-12-04:
Plus there is code already in the multicopter position controller to deal with the altitude reset.

Stifael, 2017-12-04:
it only updates the reference if arming state is stand by:
https://github.com/PX4/Firmware/commit/f0e8abe783fa8fe9f7c73be89155f4b1ccb19f81#diff-fc77c3ef569029d45764664c75d4b0c1R865

Stifael, 2017-12-04:
please also read the comment https://github.com/PX4/Firmware/commit/f0e8abe783fa8fe9f7c73be89155f4b1ccb19f81#diff-fc77c3ef569029d45764664c75d4b0c1R861

priseborough, 2017-12-04:
We are dealing with the both the handling of 'home' and 'EKF origin' which have different uses.

The EKF origin's primary purpose is to enable NED local position data to be converted to global WGS-84 coordinates. The primary purpose of home is to provide a location for RTL recovery. 

If the vehicle takes off before the EKF has started using GPS, the EKF WGS-84 origin will be set at the horizontal position where the vehicle was when EKF internal GPS checks passed. The vertical position of the origin will be set to the estimated height of the takeoff point by subtracting distance above origin from the GPS height. This does not involve a height reset because the EKF local vertical position does not change when the EKF origin is set. Prior to the origin being set, no global position messages are published so flight in auto mode is not possible. 

The existing code set sets the home lat,lon,alt using the vehicle global position and does not correct for vehicle height above origin. Home local coordinates are set to vehicle local position.  https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L1250-L1258. This is going to cause problems with RTL unless the local vertical position at takeoff is recorded and used to adjust the home alt for the vehicle height above origin. The other problem which cannot be worked around is that the vehicle will RTL to wherever it was when checks passed which will be unknown to the operator and is therefore unsafe. Until we introduce a method for the operator to confirm, adjust and approve the home location, we should disable RTL_STATE_LAND if home was set after takeoff.

In reference to the height datum issue, we could add a class variable _ref_alt_is_global to record if _ref_alt was set when vehicle_local_position.z_global was true  and if entering a waypoint controlled mode, reset _ref_alt to vehicle_local_position.ref_alt if (!_ref_alt_is_global && vehicle_local_position.z_global).

LorenzMeier, 2017-12-04:
RTL is locked out when home is not set. That is already dealt with here.

What you are describing as a potential fix for the Datum sounds reasonable. This is similar to what we had and I’m at loss when that got broken.

LorenzMeier, 2017-12-04:
@priseborough could you have a stab at that?

priseborough, 2017-12-04:
I'll look at it tomorrow.

priseborough, 2017-12-04:
Switching to auto in the air without a home position is not possible if the mission contains RTL commands. A couple of ways to deal with this that come to mind are:

1) Checking the mission plan on receipt for RTL components and setting a 'home_required' attribute to true if any RTL commands are present. The implications of handling new mission uploads containing RTL components would need to be considered.

2) Include the home location in the mission plan if operation without a home location is required. This is a better option IMO

@Stifael Can you please describe the steps you  used to reproduce this behaviour on SITL. I have not been able to do so.

priseborough, 2017-12-05:
To get option 2) to work we will need to add support for  MAV_CMD_DO_SET_HOME

The uploaded home location height will then need to be adjusted to be consistent with the takeoff height when vehicle_local_position.z_global goes true.

dagar, 2017-12-05:
Either way there are navigator changes with mission feasibility (including geofence), but I can help with that if needed.

I think the harder part (potentially) is deciding if a home position is to be required and then the interaction with the ground control station to require setting one. This includes implications for invalidating a mission based on a home position change.

Stifael, 2017-12-05:
Does it have to be that complicated? 
Commander already takes care for not valid home position by not allowing to switch into rtl (https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L449).
For auto modes where no home position is required, I think it is only required to set the relative altitude to takeoff differently. 

The way to reproduce:
- start sitl
- arm in manual or altitude mode, but before there is a gps lock
- fly around
- switch to hold

priseborough, 2017-12-05:
The easiest way initially would be to include it as part of the mission plan if takeoff without GPS is required. I see that QGC allows a home location waypoint to be inserted at the start of a mission sequence with an absolute altitude attribute, however it assigns it a command ID of 95 which is incorrect as it should be 179 according to the mavlink enumerates.

Stifael, 2017-12-05:
a takeoff without gps (for instance flow) is already possible (https://github.com/PX4/Firmware/blob/master/src/modules/navigator/takeoff.cpp#L110-L114).
Again, I don't understand yet why we need to adjust the mission plan for that

Stifael, 2017-12-05:
And I also think those are two different problems: 
1.) auto related things without gps
2.) auto related things with gps but lock occurs after takeoff

priseborough, 2017-12-05:
@Stifael I don't perceive this  as being as simple as just fixing a height offset. Getting auto to work safely without a home location is going to require hacking of the mission plan checking, geofence and RTL behaviours. 

Maybe I am underestimating the work associated with setting a home location externally, but after a couple of hours looking through the code and trying hacks to work around the mission checking and geofencing, I perceived the likelihood of introducing defects was higher compared to setting a home location before takeoff using the mission plan or GCS and adjusting the home location height when global position data is available.

priseborough, 2017-12-05:
@Stifael I agree that selecting auto without global position should not be possible. I was not able to achieve that with my testing and the log provided with this issue does not have pre-arm logging enabled.

priseborough, 2017-12-05:
What is important is the global vs local position validity, not whether there is GPS

LorenzMeier, 2017-12-05:
I think the discussion veered off a bit. What we need to fix is very basic: Set and re-set a local altitude offset first on arming and then when GPS lock becomes available.

That is how we've done it for years and what has reliably been working. This has nothing to do at all with home and nothing with the EKF's origin. It has all to do with baro vs. global altitude.

Stifael, 2017-12-05:
Thats what I meant

priseborough, 2017-12-05:
'Taking off without GPS but switching to auto in air has to be supported.'

As ofr the altitude issue,  I have a patch that in theory looks after the altitude reference change, but do not know if it solves the problem because with my SITL testing using v1.7.0-rc3 I am unable to replicate the behaviour reported.

priseborough, 2017-12-05:
OK, I can reproduce switching to hold mode after taking off without GPs and truing it on in flight

priseborough, 2017-12-05:
Including the patch from https://github.com/priseborough/Firmware/commits/pr-refAltReset stops the climb when HOLD mode is selected.

Stifael, 2017-12-05:
I quickly tested @priseborough patch. 
- switching to hold works
- waypoint maneuver works
- altitude change does not work -> will increase in altitude even if demanding negative altitude change

Mission, rtl etc are rejected. Demanding mission, a message is shown which indicates that the vehicle is not ready. Demanding rtl, negative tone alarm appears, but no other message is printed (hence a user might not even know why rtl is not possible or even worse, does not know that rtl is not possible at all).

Before hacking something together, it is worth to think once more about the benefit of supporting some of the auto modes without having gps lock available before takeoff given the CURRENT firmware state. If allowed, then we should also indicate it very clearly to the user that only a few auto related modes are possible and that rtl is not among them.

priseborough, 2017-12-05:
I got the following QGC message when I tried to change altitude from QGC after switching to hold (loiter) mode and commanding a height change.

[image]

priseborough, 2017-12-05:
It looks like this is the source of the inconsistent QGC reported height: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L1617

priseborough, 2017-12-05:
This PR should fix the height gain bug https://github.com/PX4/Firmware/pull/8418

The additional changes required to enable use of other AUTO modes will need to be postponed to until after the v1.7 release. I did find a handler for VEHICLE_CMD_DO_SET_HOME in commander.cpp

priseborough, 2017-12-05:
@DonLakeFlyer Does QGC currently support sending a VEHICLE_CMD_DO_SET_HOME message? I can see where it can be inserted as part of a mission sequence, but not an option to send it as a vehicle command.

LorenzMeier, 2017-12-07:
Fixed - please re-test master.

closed 2017-12-07 by LorenzMeier