PX4/PX4-Autopilot #8401: ECL__Lib EKF stopping navigation -> Meaning?

tops4u, 2017-12-02:
**Bug Report**

In various Logs I have seen this message, but I don't actually know what it means and what its implications are. Could you clarify the meaning and probably change the Message to something more understandable?

[image]

dagar, 2017-12-02:
At the beginning of the log it's harmless (and needs to be fixed).
ECL issue created here. https://github.com/PX4/ecl/issues/363

dagar, 2017-12-02:
@priseborough FYI

tops4u, 2017-12-02:
@dagar sorry for the duplicate, I was looking for the message in the Code and was not aware that lib ecl was a subproject of PX4. Shall I close this?

dagar, 2017-12-02:
Not a problem. We can leave it open for tracking because even after ECL is updated the reference in Firmware needs to be updated as well.

priseborough, 2017-12-03:
Will be fixed when https://github.com/PX4/Firmware/pull/8405 is merged.

LorenzMeier, 2017-12-06:
Removed that. Thanks!

closed 2017-12-06 by LorenzMeier