PX4/PX4-Autopilot #8403: ekf2 always output "EKF gps hgt timeout - reset to GPS"

garlinplus, 2017-12-03:
When I arm my pane, and switch it to altitude mode.I find when GPS signal lost, the data link immediately lost,and the rc can not to switch mode.on this moment, the console always output "EKF gps hgt timeout - reset to GPS",and the console can not receive any character and command.I guess ekf2 thread block another threads.But I read the ekf code, only the main ekf2 thread will execute sleep and poll function, and the other code in ecl library only finish algorithm.I really can not think why this.

dagar, 2017-12-03:
Which version of PX4? Can you share the flight log?

garlinplus, 2017-12-04:
version 1.50

priseborough, 2017-12-11:
Please provide a link to your log or else we have not enough information to investigate and will have to close this issue. Also please try more recent stable version. 1.5.0 is out of date.

ChristophTobler, 2017-12-27:
@garlinplus Feel free to reopen and provide logs if you face the same issue on a more recent PX4 version

closed 2017-12-27 by ChristophTobler