PX4/PX4-Autopilot #8404: vtol attitude and position uorb topic advertise problem

garlinplus, 2017-12-03:
In vtol plane,the attitude and position message should be need for mc and fw attitude control and position control.But I find the attitude and position control topic are advertise and subscribe with orb_advertise function and orb_subscribe function.I read the code of uorb, and find when multiple subscriber subscribe the same topic without orb_subscribe_multi,the subscribers all read data from the same data address.This will result bad continuity of the attitude and position.Except for it,  the same problem of uorb is not thought about.

dagar, 2017-12-03:
Could you be more specific? When vtol is enabled they're mostly using the virtual topics.

garlinplus, 2017-12-04:
For example, the data source of topic ORB_ID(vehicle_global_position) is only ekf2 or lpe or ina module.
But You can find that many modules subscribe the topic.In the driver of uorb, you can find that the kernel of uorb is a ring buffer.If there is one publisher, there only can be one subscriber.Otherwise, the data subscribed may be different of each. If the subscribing module is only use the state data,maybe there is not too big influence.But if the module have to use the data to calculate something,such as attitude control and position control,the data in one ring buffer read by many module must result bad continuity.And if there is a  bad real  time performance, the important maybe lost the important data which is read by another modules,such as mavlink, commander and vtol module.

RomanBapst, 2018-02-13:
@bkueng Sounds like something you should have a look at?

bkueng, 2018-02-13:
I'm not sure I understand you correctly. But it is certainly possible that many different modules subscribe to the same topic, and read all the same data, without dropping messages.
Of course at some point you will hit the limit where the CPU is saturated and the execution cannot keep up, and some modules drop data. But you always have this problem, and we run different modules at different priorities, such that a controller for example does not drop data.
Does that answer your question?

closed 2018-03-16 by PX4BuildBot