PX4/PX4-Autopilot #8408: MC: LowBat RTL Performed at 5m Alt instead of 50m!

tops4u, 2017-12-04:
**Bug Report** 

I was doing some testing, when I ran into a LowBattery RTL. But instead of climbing to the Preset Alt of 50m in such a case, the Copter just turned face to me an returned at roughly 5m Altitude! 

Log here: https://logs.px4.io/plot_app?log=d55046af-324a-4e6e-91ee-205b1f34bfc3

Landing was smooth an luckily the Street lantern was about 5m away.

dagar, 2017-12-04:
That's for reporting in detail. This was a bug in master (briefly) and it looks like the fix was only a few commits after the one you flew.
https://github.com/PX4/Firmware/commit/98cbd445261238f5dfc0c036e0058b2a4508b7bc

tops4u, 2017-12-05:
@Dagar :Thanks. Time for an Update I think.

closed 2017-12-05 by tops4u