PX4/PX4-Autopilot #8413: v1.7-rc3 'hangs' for split second after arm/disarm

bartslinger, 2017-12-05:
I am trying v1.7-rc3 which I flashed using QGroundControl by selecting the 'beta' option. It is a bare bones pixhawk, with an external HMC5883L compass connected to I2C. I selected airframe "Quadrotor + Tailsitter"

The brief hanging is most notable on disarm. The mavlink communication (over USB) is briefly suspended. Also notice the MAG0 Fail TOUT.

[image]

dagar, 2017-12-05:
How reproducible? Could you enable logging from boot to disarm and repeat? (SDLOG_MODE)

My first guess is the land detector updating the flight time (saved across 2 params).

bartslinger, 2017-12-05:
Lol this is getting weirder. I changed SDLOG_MODE to "from boot to disarm". It doesn't hang on arm. If I disarm the first time, time goes faster in mavlink analyze tool. Other arm/disarm sequences following that don't have a hanging moment anymore. Haven't seen any Mag0 TOUTs with this parameter configuration. https://www.youtube.com/watch?v=egteCOn_hlY

With SDLOG_MODE on default, it happens every single arm/disarm cycle. The Mag0 TOUT doesn't always pop up.

dagar, 2017-12-05:
Right, the logger itself. When logging actually begins it allocates a fairly large buffer, opens the file, writes parameters, perf, etc. When it stops it writes the perf update again.

dagar, 2017-12-05:
Might not be the main culprit here, but param save should be improved regardless. https://github.com/PX4/Firmware/issues/8294

dagar, 2017-12-05:
For the time issues in mavlink it's likely the main logger task which has a higher priority than any of the mavlink threads.

bartslinger, 2017-12-05:
But it should not have higher priority than the compass, right? Because I did get a Mag0 TOUT error message sometimes.

dagar, 2017-12-05:
External I2C mags run in the high priority work queue (HPWORK), but so does the land detector. 

Can you share the log that corresponds with your youtube video (or similar)? We can check the perf counters and logged cpu utilization.

tops4u, 2017-12-05:
I can confirm that I have seen MAG0 TOUT occasionally at Disarm (can't tell if just before, during or after). 

Here a Log, showing the phenomenon (but here also my IOProc was blinking red for some strange reason - could this be related)?

https://review.px4.io/plot_app?log=89bd04af-1f92-4b8e-ab21-6dde1aab04ee

LorenzMeier, 2017-12-06:
@bkueng Could you please look into this? I think the param save after disarming is too disconcerting in the logs and we need to find a better approach.

bkueng, 2017-12-06:
I'll take a look. If urgent, we can increase the mag timeout (until we find something better), it is the lowest-risk change for the release.

bkueng, 2017-12-07:
Here's what I see from the log & code:
- No CPU spikes. But if some proces was busy/hogging the CPU for the amount of time needed to create the mag timeout (which is set to 300ms), I would expect to see a CPU spike in the log.
- all modules (commander, land detector & ekf2) that potentially change params after disarm do not notify the system
- The provided log shows that the landing detection happens after disarming (it's not in the log anymore, but it needs to be at least several 100ms later). ekf2 and commander potentially set params on disarming, but the land detector sets them when landing is detected. Which means that if land detection is several 100ms after disarm, it can coincide with an ongoing param save and thus will block.

--> The land detector (and thus HPWORK and the mag) got blocked trying to set the flight time param, because a param save was in progress.
PR: #8428

tops4u, 2017-12-07:
@bkueng I had to switch to Stab and Disarm, as it seemed to me that Landing was not properly recognized and 2 Motors started to spin up. To prevent Flipping I decided to disarm myself.

dagar, 2017-12-07:
@bartslinger could you give master another try?

bartslinger, 2017-12-07:
Still a bit of a hang visible in the mavlink comms during arm/disarm, when logging from arm till disarm. I haven't seen the Mag0 fail TOUT message on master.

dagar, 2017-12-07:
The mag0 timeout is the larger concern. The hang in mavlink I consider more of an annoyance, but can also be improved (likely after v1.7.0.). The logger start and stop could be pushed into the low priority writer thread. https://github.com/PX4/Firmware/blob/master/src/modules/logger/logger.cpp#L1478

Thanks @bkueng!

bartslinger, 2017-12-08:
Alright, I don't really care about the mavlink delay either.

closed 2017-12-08 by bartslinger