PX4/PX4-Autopilot #8416: Geofence Max Alt Logic Too Abrupt, and does not protect user from commanding bad alt

raygamero, 2017-12-05:
(This was discovered on a recent build on the Gazebo SIM)
User is able (without warning or lock-out) to command altitudes that are the same as the geofence max altitude.  
For example.  The Geofence was set to 300 ft (terminate) and the mission takeoff altitude was set to 300 ft.  The aircraft climbed to 300 ft precisely and then flight terminated with the message "Maximum altitude above home exceeded by 0.0m"

I understand that there are safeguards when trying to command ABOVE the max altitude, setting the geofence altitude to your ceiling and then commanding the aircraft to the same ceiling is an understandable mistake.   Can we safeguard this edge case further?

1. Don't let a valid altitude command trigger the failsafe  (make the trigger at > alt instead of =).
2. Provide warnings when commanding close to or approaching max.
3. Provide a small buffer zone or trigger when aircraft vector is exceeding max.

dagar, 2017-12-05:
TODO

 - include the acceptance radius when checking the mission geofence validity.
 - geofence warn before breach (acceptance radius again?)
 - future work to consider - factor in vehicle velocity to retake control and prevent actual breach

santiago3dr, 2018-08-10:
has altitude acceptance logic been changed?

closed 2019-02-19 by stale[bot]