PX4/PX4-Autopilot #8419: Release v1.7 sign-off

mrpollo, 2017-12-06:
Is your team ready for a release?

* [ ] Project / Founder - @LorenzMeier
* [ ] Communication Architecture - @bkueng , @julianoes 
* [ ] UI / UX - @dogmaphobic, @DonLakeFlyer  
* [ ] Multicopter Flight Control - @Stifael, @MaEtUgR
* [ ] VTOL Flight Control - @tumbili, @AndreasAntener, @sanderux @bresch
    * [ ] VTOL standard
    * [ ] VTOL tailsitter
    * [ ] VTOL tiltrotor
* [ ] Fixed Wing Flight Control - @dagar, @priseborough 
* [ ] Racers - @anton-matosov 
* [ ] OS / drivers - @davids5 
* [ ] UAVCAN / Industrial - @pavel-kirienko 
* [ ] State Estimation - @jgoppert, @priseborough 
* [ ] VIO - @mhkabir, @ChristophTobler 
* [ ] ~Obstacle Avoidance - @vilhjalmur89~ - not implemented in mainline codebase yet.
* [ ] Snapdragon - @mcharleb 
* [ ] Intel Aero - @lucasdemarchi, @simonegu 
* [ ] Raspberry Pi / Navio - @bkueng 
* [ ] Parrot Bebop - @eyeam3 
* [ ] Airmind MindPX / MindRacer - @iZhangHui 
* [ ] Test team - @d3z0rian, @mrpollo

mrpollo, 2017-12-06:
Here's the v1.6 release sign-off issue https://github.com/PX4/Firmware/issues/7258

sanderux, 2017-12-06:
Can we get a final release candidate for testing?

iZhangHui, 2017-12-11:
MindPX/MindRacer pass our flight testing, and we are ready for v1.7 release.

mrpollo, 2017-12-16:
v1.7.0 was released, thanks to everyone here for their help pushing this one out

https://github.com/PX4/Firmware/releases/tag/v1.7.0

closed 2017-12-16 by mrpollo