PX4/PX4-Autopilot #8421: Changing data rate in pixracer (px4 1.6.5)

Changliu52, 2017-12-06:
I am trying to increase imu data rate in pixracer and output through maros.
I tried to add the directory /etc/ in the sd card, and then added the file extra.txt with the content:

```
#!nsh
#
# USB MAVLink start
#

mavlink start -d /dev/ttyACM0 -b 921600
# Enable a number of interesting streams we want via USB
mavlink stream -d /dev/ttyACM0 -s PARAM_VALUE -r 300
mavlink stream -d /dev/ttyACM0 -s MISSION_ITEM -r 50
mavlink stream -d /dev/ttyACM0 -s NAMED_VALUE_FLOAT -r 10
mavlink stream -d /dev/ttyACM0 -s OPTICAL_FLOW_RAD -r 10
mavlink stream -d /dev/ttyACM0 -s VFR_HUD -r 20
mavlink stream -d /dev/ttyACM0 -s ATTITUDE -r 20
mavlink stream -d /dev/ttyACM0 -s ACTUATOR_CONTROL_TARGET0 -r 30
mavlink stream -d /dev/ttyACM0 -s RC_CHANNELS_RAW -r 5
mavlink stream -d /dev/ttyACM0 -s SERVO_OUTPUT_RAW_0 -r 20
mavlink stream -d /dev/ttyACM0 -s POSITION_TARGET_GLOBAL_INT -r 10
mavlink stream -d /dev/ttyACM0 -s LOCAL_POSITION_NED -r 30
mavlink stream -d /dev/ttyACM0 -s MANUAL_CONTROL -r 5
mavlink stream -d /dev/ttyACM0 -s HIGHRES_IMU -r 200
mavlink stream -d /dev/ttyACM0 -s GPS_RAW_INT -r 20

# Exit shell to make it available to MAVLink
exit
```
I then tested it through direct usb connection. But from mavros, the mavros/imu/data_raw is still published at 50 hz.
What did I do wrong?

Chang

dagar, 2017-12-06:
The file is `etc/extras.txt` (note the s).

Changliu52, 2017-12-06:
Thanks for the reply. 
Sorry. That was a typo in my question. I did have extras.txt in the /etc directory in my sd card.

Also I noticed that my sd card was completely empty. Is this normal? Or are there supposed to be some default files to edit rather than creating new files manually? 

The pixracer doesn’t complain anything, but is it possible that the it has problem accessing the sd card?

Thanks
Chang

closed 2018-04-29 by Changliu52