PX4/PX4-Autopilot #8453: make posix_sitl_default gazebo - sonarSens.pb.cc

FaboNo, 2017-12-12:
I am trying to build the gazebo SITL simulator with make posix_sitl_default gazebo

I ended up with the following error:
no rule to build « /home/fabrice/src/Firmware/Tools/sitl_gazebo/PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND », required for « sonarSens.pb.cc

I am a bit puzzled because I cannot find sonarSens.pb.cc and the protobuflib.dev is already installed.

Somebody is facing the same issue?

Thank yo

Fabrice

ChristophTobler, 2017-12-27:
I can't reproduce this error.
Do you have the following packages installed: `libprotobuf-dev` and `protobuf-compiler`? (not sure what else it could be...)
Also, try (if you haven't already) `make clean`, `git submodule update --init --recursive` , `make posix_sitl_default gazebo`

FaboNo, 2018-01-04:
@ChristophTobler thank you, actually libprotobuf-dev was missing!

closed 2018-01-04 by ChristophTobler