PX4/PX4-Autopilot #8454: FW: Bug when switching from Stabilized to Altitude mode

acfloria, 2017-12-12:
When switching from stabilized mode to the altitude hold mode for a fixed-wing while tracking a non-zero attitude setpoint the airplane is still tracking that last attitude setpoint instead of 0 roll/pitch.

I think the reason is that in stabilized mode the attitude controller does [generate a setpoint by itself](https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/fw_att_control_main.cpp#L980). However, in the altitude hold mode this is not the case anymore and I guess that the l1 controller also does not generate a setpoint so that the last setpoint from the stabilized mode is tracked.

dagar, 2017-12-12:
Do you have a log showing this? There should be a new attitude setpoint from the FW position controller. As a side note I'd like to get the setpoint generation completely out of the attitude/rate controllers.

acfloria, 2017-12-13:
I actually just noticed it during our pre-flight tests so I do not have a log file now. But I will go flying today and then I will test it again in flight.

acfloria, 2017-12-14:
I did two tests, one in the [lab](https://logs.px4.io/plot_app?log=b1ee527f-cd8a-462c-bac0-55d470899abb) and one [test flight](https://review.px4.io/plot_app?log=9be3df0d-c3b7-4fae-b4d4-fa1d3fea0627).

The issue only appeared during the in lab test. For example between 0:42-0:53 when I change the throttle/pitch/roll input but the output is not changed.
In the lab I do not have a GPS signal, could this be an issue?

closed 2018-08-10 by santiago3dr