PX4/PX4-Autopilot #8457: Option to disable magnetometer during preflight check

yashm, 2017-12-12:
**Feature Request**
Add option to disable magnetometer for preflight check.

Posting as a feature request per @dagar's suggestion.

Use cases:
1. To reduce errors while flying in cluttered indoor environments.
2. For Pixhawks that don't have mag hardware

dagar, 2017-12-12:
It's easy enough to add a parameter to make commander tolerate this (skip the checks), but I'm wondering about the estimators.

yashm, 2017-12-12:
The estimator does have an option to ignore the magnetometer during sensor fusion under the param:
`EKF2_MAG_TYPE`.
But the absence of having the mag on board altogether fails during the preflight check (which makes sense in the current SW stack).

dagar, 2017-12-12:
Let me take a look at how the preflight check code is organized. Maybe we could add something like a single parameter that's a bitmask with all of the possible checks. That would help you here and cover all kinds of future special use cases.

yashm, 2018-01-02:
Is there any quick hack to test this for the time being?
Comment out some checks in the source code perhaps?

I'd like to test the first batch of PCBs till PX4 v1.8 is released.

dagar, 2018-01-04:
You could hack it out here. https://github.com/PX4/Firmware/blob/master/src/modules/commander/PreflightCheck.cpp#L646

closed 2018-08-10 by santiago3dr