PX4/PX4-Autopilot #8459: Unable to get position lock in v1.7.0-rc4 on Pixhawk Mini

myeabhishek, 2017-12-13:
I have a vehicle with quad x configuration with Pixhawk mini as the flight controller. I installed v1.7.0-rc4 on my Pixhawk mini. I was getting various different issues related to position lock. The issues are described below:

1) 12 Satellites received. Green light glowing, however, auto takeoff denied. In flight view the drone was constantly changing position on GUI.
2) On reinstalling same firmware, 0 satellites received and no position lock. Green light didn't glow.
3) On reinstalling same firmware again, 11 satellites received, however, green light didn't glow and autonomous takeoff denied.

I tried changing Pixhawk mini and GPS but the issue still remained. 

At last I reverted to 1.6.5 version and then everything worked seamlessly. I got a GPS lock and I was able to give an autonomous mission as well.

What could be the issue?

dagar, 2017-12-13:
Could you post the log? https://logs.px4.io/

myeabhishek, 2017-12-13:
I don't have the logs as I was not able to arm at all in autonomous mode.

dagar, 2017-12-13:
Set `SDLOG_MODE 2` to enable logging from boot until disarm. Then we'll be able to see exactly what's happening in the firmware.

Having a GPS lock isn't enough. The position estimate (from GPS and other sensors) error must be sufficient for autonomous flight. We definitely need to improve the error messages and feedback in this area.

Please try it again with the latest development PX4 and logging from boot (SDLOG_MODE) and post your log https://logs.px4.io/. From that I'll be able to tell you for sure what's happening.

myeabhishek, 2017-12-15:
@dagar I generated the logs in the required setting as suggested by you. I was able to get 11 satellites, however I didn't get a position lock. Please find the attached logs. I was using v1.7.0-rc4.

I have uploaded the logs as well. Please find them at the attached link.

[log_38_2017-12-15-16-48-06.ulg.zip](https://github.com/PX4/Firmware/files/1562729/log_38_2017-12-15-16-48-06.ulg.zip)

closed 2018-05-10 by PX4BuildBot