PX4/PX4-Autopilot #8462: sensors status ideas

dagar, 2017-12-13:
Piecing together sensor issues from logs is often more difficult than it needs to be. I was thinking of expanding `sensors status` to include the device id (decoded), rotation, internal/external, mapping between uORB numbering and CAL parameters, etc.

Then sensors status could be appended to the log preflight and postflight just as as perf and top are now.

Alternatively we publishing a few of the key details that are missing and do more work on flight review to decode and summarize. A snapshot of the sensors status table can also be helpful for debugging sensor rotation issues.

@davids5 @priseborough @bkueng

bkueng, 2017-12-14:
Sensor status identification can definitely be improved and these are some good ideas. `sensors status` is a good place for that.
Adding that directly to the log has the disadvantage that it won't work on archs where logger runs on a different CPU, such as snappy (there might be others in future) - perf & top already have that problem, so we might have to find a more generic solution. Such as logger sending a shell command to the FMU and receiving its stdout.

dagar, 2017-12-14:
Thanks, that's a good point.

closed 2018-08-10 by santiago3dr