PX4/PX4-Autopilot #8463: airspeed "quick cal" advanced option

dagar, 2017-12-13:
Airspeed calibration determines the offset of the differential pressure sensor averaged over a few seconds (# samples), then applies the offset, waits briefly, then asks you to blow to verify the tubes are connected properly.

This should definitely remain the default, but more advanced users would like the option to zero the differential pressure sensor immediately and skip the blowing step.

This could be an advanced param for airspeed quick calibration.

@raygamero

raygamero, 2017-12-15:
Thanks for this issue Daniel.  I'm going to see if I can come up with a good solution.  
Do you mind telling me where to start to understand the current cal module?

Antiheavy, 2017-12-15:
Interesting idea.  Most other autopilot systems don’t have a diff pressure test step although I really like it.  Actually I would love to see it as a pre-arming check as well, but I have thought through how that would be implemented firmware side. 
Personally, I would like to see the averaging time cut in half to speed up calibration. The time now seems too long and may introduce more chances for error than it tries to average out. As mentioned, I really like the “blowing” step to verify the pneumatics. ...although I never blow into a Pitot (moisture from mouth/lungs can be bad news) , I press my finger over the end to create and hold pressure.

tubeme, 2018-01-14:
Hi Daniel, that is a very good idea. I understand the procedure of calibrating the airspeed sensor has two aspects. 1. The most important to see the hardware inbuilt difference between the two tubes ie the two sensors in the chip at same given pressure (that is why we need not to fly and cover the pitot at that moment). 2. To see what is the current reading at static pressure and take it as zero.

This complicates the whole use case because we cannot get this two different calibrations separatelly when we need them.

As far as I understand the pressure difference in reading is something that not change with time, because it is inbuilt the sensors. Or maybe it changes with time but very very slowly. So it does not matter if it is done in closed space in the workshop. This can include blowing the tube.

The second calibration can be done on the airfield and it definitely should not include blowing in the tube. It just has to get the reading from the static tube and take it as zero.

I don't know if by quick cal you mean this? 

But using airspeed now is so cumbersome it is not real. Especially on a windy day on the field...

raygamero, 2018-01-16:
I'm intimately aware of high dollar UAS that, for day to day flying, have a procedure to "bag" the pitot (placing a loosely fitting bag over the tube to block wind) and then "zeroing" the sensor.  This is simpler and safer, assuming that you don't leave the bag on the pitot!  This would be a "quick cal" or some descriptive title. 
The initial check, to verify if your tubes are not swapped is great and should not be removed.  It is just not used every time.  Maybe just after repairs or during inspections.

Antiheavy, 2018-01-16:
> This could be an advanced param for airspeed quick calibration.

instead of a param to switch what the routine does, should it be two different routines that can be called separately?  I realize QGC (or your favorite software) would then have to deal with two different message sets, but the "zeroing" and "testing" really are two different things.

I've got the reverse argument of some others in the thread.  There are newer differential pressure sensors that claim to never need in-field calibration (I'm skeptical, but would happily be proved wrong) - I still want the ability for a "test" of the pitot-static tube even if the calibration routine isn't run.  Not only are we testing that the tubes are not backwards, but we are testing for leaks and blockages that easily occur when landing in rough ground.   A future firmware feature might be an option to require a successful diff pressure "Test" to unlock arming.

Antiheavy, 2019-01-29:
this is still relevant.

Antiheavy, 2019-06-25:
still relevant.  @kjkinney I wonder if this is something we can look at?

Antiheavy, 2019-10-03:
This "quick cal" feature would also be neat to be able to run at boot.  either as a optional parameter controlled thing, or maybe scripted as part of the start-up scripts.

dagar, 2019-10-03:
> This "quick cal" feature would also be neat to be able to run at boot. either as a optional parameter controlled thing, or maybe scripted as part of the start-up scripts.

I wonder if we could get away with doing it automatically if disarmed and any time the differential_pressure is sufficiently stable for a relatively long period of time. Maybe combined with the vehicle being still (attitude and position)?

Antiheavy, 2019-10-03:
>  sufficiently stable for a relatively long period of time

some of the cheaper (and more popular) diff pressure sensors are never really stable, especially at the very low end of the scale (very noisy and bouncy).

Also, having it happen at boot up is something that can easily be explained/training.  e.g. one option to recal airspeed is to just reboot the vehicle and block the pitot from wind.