PX4/PX4-Autopilot #8464: Not getting MAV_LANDED_STATE_ON_GROUND

lovettchris, 2017-12-14:
See [flight log](https://review.px4.io/plot_app?log=4d7df5bf-d8b5-43a5-bd9a-842f8702a131).  I expect to get extended sys state showing MAV_LANDED_STATE_ON_GROUND after doing simple take off and landing.

**Repro**
1) Run AirSim, with PX4 connected, either SITL or HITL mode.
2) Run this python script 

```python
from AirSimClient import *
import sys
import time
import argparse

client = MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)

client.armDisarm(True)

landed = client.getLandedState()
if landed == LandedState.Landed:
    print("taking off...")
    client.takeoff()
        
time.sleep(2)

print("landing...")
client.land()

print("disarming.")
client.armDisarm(False)
```

Notice client.land() times out because the extended sys state is stuck in MAV_LANDED_STATE_LANDING, and never reaches MAV_LANDED_STATE_ON_GROUND.  This seems to be a regression, older versions of the firmware worked fine.

dagar, 2017-12-14:
Do you recall the last working version? Was it from PX4 master or a stable release?

lovettchris, 2017-12-15:
I tried [1.6.5](https://github.com/PX4/Firmware/releases/tag/v1.6.5) and same issue.  I couldn't find available px4 builds for earlier releases to test, but I know for sure it was working on 1.5.5.

lovettchris, 2017-12-17:
So I tested 1.6.4, 1.6.0 and same behavior, so I went back to 1.5.5 and now I'm not sure if it is doing the right thing either.  What I found is that I have to "disarm" the drone in order to get the "landing detected" message, and the associated MAV_LANDED_STATE_ON_GROUND landed state in extendedsys_state, but that is silly because I need to get "landing detected" before I can safely disarm.

dagar, 2017-12-17:
This is likely something to do with the land detector in simulation.

What's your typical Airsim setup? HIL, SITL, airframe, etc?

dagar, 2017-12-17:
[image]

You can see it was disarming that forced the landing state change.

[image]

lovettchris, 2017-12-17:
I think I found the problem, AirSim is not simulating a ground hit positive acceleration.  It does simulate the correct larger negative acceleration on takeoff:
[image]

but the landing is too smooth, so the PX4 landing detector didn't see it.

[image]

Would this explain the problem?

LorenzMeier, 2017-12-17:
That might be an issue - also if you set a too low landing speed. You need to have at least 0.5 m/s, otherwise the system can't really judge if the speed is changing on contact.

closed 2019-01-29 by lovettchris