PX4/PX4-Autopilot #8465: It is possible to arm to mission with stale baro

CarlOlsson, 2017-12-14:
**Bug Report**
I flashed master cbdb08bb6109c1ea3ef4e23761233df447523a52 on to a pixracer quad, with the only difference that the baro altitude was overwritten to -15 km in the [sensor driver](https://github.com/PX4/Firmware/blob/cbdb08bb6109c1ea3ef4e23761233df447523a52/src/drivers/ms5611/ms5611.cpp#L868-L868).
```
report.altitude = (((pow((p / p1), (-(a * R) / g))) * T1) - T1) / a;
report.altitude = -15000.0f;
```
When connecting to telemetry I get the warning that the baro data is stale
```
0:00:05 ERROR: [sensors] Baro #0 fail:  STALE!
```
Then I upload a mission in qgc and try to start it, I get as expected the messages about failing preflight checks
```
0:00:21 WARNING: [commander] REJECT AUTO MISSION
0:00:45 WARNING: [commander_tests] PREFLIGHT FAIL: EKF INTERNAL CHECKS
0:00:45 WARNING: [commander] Arming not possible in this state
```
However, after that the quad is **armed and tries to fly away**
At 49 s the quad is armed in mission mode and all motors start, at 52.3 s I switch to assisted so I can unplug the battery more safely.  
[image]

Log is [here](https://logs.px4.io/plot_app?log=fcdc27bc-705d-4ef6-80d0-efeadd82c219)

dagar, 2017-12-14:
Maybe we should do a top down review of barometric data in the system. There are lots of little side issues.

 - QNH usage
 - calibration
 - pressure altitude calculation (https://github.com/PX4/Firmware/pull/8466)
 - ms5611 jumps (https://github.com/PX4/Firmware/issues/7)
 - commander baro checks

Antiheavy, 2019-01-29:
> Maybe we should do a top down review of barometric data in the system. There are lots of little side issues.
> 
>     * QNH usage (#8253) ([#8251 (comment)](https://github.com/PX4/Firmware/pull/8251#issuecomment-342628114))
> 
>     * calibration
> 
>     * pressure altitude calculation (#8466)
> 
>     * ms5611 jumps (#7)
> 
>     * commander baro checks

this is still a relevant list.

CarlOlsson, 2019-06-25:
I would keep this open since it has not been explicitly addressed

julianoes, 2020-01-30:
FYI @MaEtUgR relevant regarding preflight checks.

kaklik, 2022-07-18:
That issue seems to be no longer valid for the current firmware versions, probably should be closed.

julianoes, 2022-07-18:
Thanks @kaklik

closed 2022-07-18 by julianoes