PX4/PX4-Autopilot #8468: home position update at mission start

sanderux, 2017-12-14:
A corner case was uncovered in the following scenario:

- Takeoff from A at altitude 0
- Land at B at altitude 100
- Update the mission to return, setting a vtol_takeoff at 50m relative to home
(no restarting of FC)

The vehicle proceeded to arm and calculate the vtol_takeoff condition but did so before the home position was reset. Altitude condition was met and it proceeded to transition 0.5m above ground. (crashed into a fence)

Discussed with @dagar the following additions required:

Possible fixes:
- wait for home position to update before navigator starts
- sanity check vtol_takeoff (keep in mind it IS possible this wp is called when in air)
- continuously update home position when landed && disarmed

sanderux, 2017-12-16:
@priseborough @LorenzMeier FYI

sanderux, 2017-12-16:
Option for continuous updates while landed & disarmed:
https://github.com/PX4/Firmware/blob/b5be990109a00814102005b6b809d64daadc40d8/src/modules/commander/commander.cpp#L3123-L3141

Option for home position update at disarm:
https://github.com/PX4/Firmware/blob/b5be990109a00814102005b6b809d64daadc40d8/src/modules/commander/commander.cpp#L736

Option for additional checks before loading new mission:
https://github.com/PX4/Firmware/blob/b5be990109a00814102005b6b809d64daadc40d8/src/modules/navigator/mission.cpp#L84

sanderux, 2017-12-16:
A possible issue that made this happen:
The following only runs on_inactive for mission. If the vehicle never leaves the mission state while updating the mission this will never run:
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L104

sanderux, 2017-12-18:
Proposed fix for 1.7:

In home_position.msg:
Add bool manual_home = false

In commander:
Set bool manual_home = true when home set manually

In commander: 
if disarmed && landed && homepos.manual_home == false 
    && distance(current_pos, home_pos) > eph*2 
    && distance(current_alt, home_alt) > epv*2: 
update_home_position() 

In mission: 
If home_position_updated: 
update_off/onboard_mission()

priseborough, 2017-12-18:
The continuous update of home position whilst landed and disarmed seems like the a workable compromise. However a consequence if a geofence is active and the vehicle lands and disarms at a new location is that the geofence moves with it.

sanderux, 2017-12-18:
I think the moving of the geofence might even be desirable. I'll get working on this

dagar, 2017-12-30:
I believe this was resolve by the merged PR. @sanderux please reopen if I'm wrong.

The other outstanding piece is mission feasibility on home change, but I'll tackle that in some other related work.

closed 2017-12-30 by dagar