PX4/PX4-Autopilot #8472: Feature Request: AutoTune

mzahana, 2017-12-15:
Hi,

Is there a plan to have auto PID gain tuning like the one in APMCopter?

whyscience, 2017-12-16:
Need this feature, too.

mzahana, 2017-12-18:
Probably the autotune implementation in APMCopter can be helpful!
https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode_autotune.cpp
I might work on it on a very slow pace.

hamishwillee, 2017-12-18:
You can port the ideas, but not the code - I think they have incompatible licensing. I've seen some discussions that the tuning on APMCopter is a bit "rough" for many frames, and what would be better is an incremental tuning that happens across a longer time while flying.

TSC21, 2017-12-18:
We actually have a good tuning guide on the user guide of PX4. So I don't see why we would want an auto-tuning feature. Also, ideally we would want to evolve into a LQR capable tuning.

hamishwillee, 2017-12-19:
> We actually have a good tuning guide on the user guide of PX4. So I don't see why we would want an auto-tuning feature. 

People don't read, and sometimes they screw up. Even if I did a first level manual tuning myself I would personally love a system that continued to improve while flying.

TobiMiller, 2017-12-23:
I dont think it is the point with "nobody wants to read". it is more like -> go in air -> push a button -> get some good values for pid --> ready to go. i think it is to inconvenient to put your copter into a structure which holds it with wires and tune for hours until you get it flying good.

hamishwillee, 2017-12-24:
Fair enough.

dagar, 2017-12-24:
FYI, you can already do some of this offline. https://github.com/dronecrew/px4tools/blob/master/examples/Log%20based%20System%20Identification%20and%20Control%20Design.ipynb

@diegoeck is working on this as well.

TobiMiller, 2017-12-24:
@dagar 
thanks for that! i think something like that needs to be more announced. i knew it but i found it just by luck.

hamishwillee, 2017-12-27:
@dagar Is this anything that could/should be documented in PX4 User Guide? On quick scan I'd say "too early" - would be nice if some of this was in http://logs.px4.io though.

diegoeck, 2017-12-27:
Hi! I'm working with some of my students in a toolbox with data-driven control for PX4. I personally don't like this automatic control techniques that twitch during a fly. Our approach is to fly with some controller, get the logs (like in logs.px4.io) and design a better controller from data logs. It is close to the @jgoppert approach, but while he identify a model and design the controller based on the model, we directly design the controller from data (that's why we call data-driven). We have already tested our approach for rate and angle (pitch, roll and yaw), and now we are adapting the algorithms to work directly with the logs that px4 already collect. Since our algorithms are very simple, they could run on-line (on the website). But, as any algorithm, there are some parameters that the user needs to choose, and we are testing some default parameters that may work with many drones.
As soon we deliver something stable we will ask some people to test it. I hope people like it.

zarathustr, 2018-01-03:
you can simply write a LQR identifier to obtain the system parameters. I think there have been some examples on the github e.g. LibrePilot (TauLabs). However, the stability and safety of this variety of techniques are questionable from time to time.

mzahana, 2018-01-03:
@zarathustr Can you please share some specific references on the theory part?

mbr1989, 2018-04-30:
push

closed 2018-08-10 by santiago3dr

mzahana, 2021-02-14:
@dagar @diegoeck  @TSC21 @hamishwillee any updates on this?

I did a script for offline tuning a year ago (https://github.com/mzahana/px4_pid_tuner ), but have not really tested it thoroughly.

I was wondering if anyone else made any progress on this.
Cheers.

coderkalyan, 2021-11-13:
@mzahana https://github.com/PX4/PX4-Autopilot/pull/18311