PX4/PX4-Autopilot #8473: build failed due to c++ library

YuYangguang, 2017-12-16:
I tried to build the firmware in Ubuntu 14.04. I have installed the gcc 4.9, g++ 4.9 and build-essential, but when I tried to build the code, the system report the following error:
In file included from /home/cloudy/fangzhen/Firmware/src/lib/matrix/matrix/math.hpp:3:0,
                 from /home/cloudy/fangzhen/Firmware/src/lib/mathlib/math/Matrix.hpp:51,
                 from /home/cloudy/fangzhen/Firmware/src/lib/mathlib/mathlib.h:45,
                 from /home/cloudy/fangzhen/Firmware/src/drivers/device/integrator.h:45,
                 from /home/cloudy/fangzhen/Firmware/src/drivers/device/integrator.cpp:43:
/home/cloudy/fangzhen/Firmware/src/lib/matrix/matrix/stdlib_imports.hpp:14:21: fatal error: cinttypes: No such file or directory
 #include <cinttypes>
                     ^
compilation terminated.
make[3]: *** [src/drivers/device/CMakeFiles/drivers__device.dir/integrator.cpp.obj] Error 1
make[2]: *** [src/drivers/device/CMakeFiles/drivers__device.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
Scanning dependencies of target drivers__led

It seems that I didn't set the environment correctly. But I just didn't know how to slove it. Anyone could help me?

dagar, 2017-12-16:
Did you install all packages mentioned in the dev guide? https://dev.px4.io/en/setup/dev_env.html

LorenzMeier, 2017-12-30:
Closing as stale.

closed 2017-12-30 by LorenzMeier

liruihao, 2018-10-23:
I have encountered the same problem, have you solved it? 

Kind Regards,

hamishwillee, 2018-10-24:
@liruihao Please don't append to old closed issues - there is a good chance no one is looking.

The build works find on the supported OS - Ubuntu 16.04 - if you follow the instructions in https://dev.px4.io/en/setup/dev_env.html