PX4/PX4-Autopilot #8475: can't change to Offboard and preflight failed mag sensors inconsistent

raywang-nx, 2017-12-16:
Dear:
   here are the log files.
   https://logs.px4.io/plot_app?log=b11cdad4-8c8a-47c9-a83f-72c11df0df22
   https://logs.px4.io/plot_app?log=7d3eac38-1274-451f-9d89-d1f44462aa91
   https://logs.px4.io/plot_app?log=50c58947-0613-4c6f-b599-f814727564ca
   
    i am not sure all of these three are the right files. but at least one of them is.

    there are two questions puzzle me right now.

    first, i cannot arm pixhawk at all, it keep telling me  mag sensros inconsistent. i have placed ublox m8n with the same direction as pixhawk, and calibrate them 2 times, and i still got those msgs. what can i do?

   second, sometimes ,i can arm it as expected, but when i try to change it to the Offboard mode, it can't.
pixhawk's red light blinks for maybe 4 times, with buzzer rings 3 times, and can't change it to offboard.

i think i need to send several position commands to it before, but everything seems the same.

and someone said that after the drone fly in the sky, change to alt-hold(may be other name i guess), then change to offboard and success, but i manage to change to position mode , but fail to change to offboard with the same situation like this. neither with rc nor mavros. do i miss some steps or so?
right now i arm it, speed up the throttle(not with porps), change to altitude mode, and then try to change it to offboard, but failed.

dagar, 2017-12-16:
What's your physical setup? Where are the compasses relative to sources of high current? 
Are you doing the full 6 position compass calibration? Are you doing it away from buildings, cars, etc?

If you change logging to start at boot (SDLOG_MODE 2), log sensor comparison (SDLOG_PROFILE see screenshot), and go through the full calibration again we can look at the log to find the issue.

[image]

raywang-nx, 2017-12-18:
this morning, i calibrated it again with all the six positions, and all te necessary pre steps, now it works fine, if something wrong, i will reopen it again

raywang-nx, 2017-12-19:
@dagar 
Sorry, but it seems everything works fine just after the moment i calibrated the compass, but later after that, it comes mag inconsistent again.

here is the calibrate logging data
https://logs.px4.io/plot_app?log=0d5cd039-59e1-4e8b-a2ea-d4e85b733588

here is the later ones
https://logs.px4.io/plot_app?log=966addcc-a14b-457a-a29d-06943625438a

raywang-nx, 2017-12-19:
i have checked  https://github.com/PX4/Firmware/issues/7327 just now, then i list my dev here

nsh> ls /dev
/dev:
 accel0
 accel1
 adc0
 baro0
 console
 gyro0
 gyro1
 hmc5883_ext
 l3gd20
 led0
 lsm303d_accel
 lsm303d_mag
 mag0
 mag1
 mmcsd0
 mpu6000_accel
 mpu6000_gyro
 ms5611_spi_int
 mtdblock0
 mtdblock1
 null
 pipe0
 pipe1
 pwm_output0
 pwm_output1
 px4fmu
 px4io
 ram0
 rgbled0
 tone_alarm0
 ttyACM0
 ttyS0
 ttyS1
 ttyS2
 ttyS3
 ttyS4
 ttyS5
 ttyS6
nsh> 

i think there must be something wrong with my mag devices. so i type lsm303d start then arm pixhawk with rc, while the mag inconsisitent error disappears, so will it be such a software error? or a bug? if you need anything that i can help, just @me, i am willing to help

dagar, 2017-12-19:
The mag inconsistency check is new. You can silence it by increasing COM_ARM_MAG, however the problem is still there.

With the additionally logging enabled (all mags) can you try this.

 - point vehicle north and hold for a few seconds
 - spin slowly 360 degrees
 - hold vehicle nose down
 - spin slowly 360 degrees

Then we can look at the look and see just how different your mags are. I would guess there's a problem with how you've mounted everything. If one of the mags is definitely not usable you can disable it via param (CAL_MAGX_EN 0).

mzahana, 2017-12-19:
I am facing the same problem with the MindPX flight controller. So far, I have been doing the test indoor. I am using a customized 3D printed quadcopter where the flight controller, companion computer, and all the wiring are inside a hub and close to each other. I flew it indoor and outdoor using v1.6.5 and it flew nicely. After upgrading to v1.7, I started to get the `MAG SENSORS INCONSISTENT` many times even after calibrating, many times. MindPX has 2 internal mags and I used an external 3DR GPS with a mag. The only up to now that I was able to git rid of the error is to disable the internal mags and leave the external one. That way it worked fine indoors. I have tried it outdoors yet, but I guess it should be fine.

My conclusion is that, most probably that the internal magnetometers are highly affected by the surrounding wiring. However, it was working fine using 1.6.5 which is weird to me!

[image]

dagar, 2017-12-19:
@mzahana your mags were likely always inconsistent, you just weren't getting the error.

If you're interested you should try the logging I described above and we can see how much they actually differ. Then you can decide if the internal should be disabled (CAL_MAGX_EN) or if the the inconsistency check should be relaxed.

deksprime, 2017-12-25:
I am also gettin the exact same error. 1.6.5 I had no problems,1.7.0 now this problem comes up. Also I believe that my problem wasnt there as @dagar says, because now my compass readings have a consistent 45 degrees error. I will follow the logging steps that you have mentioned and will be sharing a log here

raywang-nx, 2017-12-27:
@dagar Sorry for being away for such a long time,  right now it seems i can't make it right by just calibrating it, i have been doing this for quite a few times, maybe it is just because i do it near my PC, luckily, by increasing the COM_ARM_MAG value, i can make it, maybe not the best way to solve this though. thank you any way, if i got something new, i will post it here again.

mebj, 2018-03-16:
I have the same problem on a Pixracer R14 with Px4 R1.7.3 stable. I have tried to recalibrate compass at different location without success. Shutting of the external compass eliminates the error "Mag sensors inconsistent". 

Also raising the COM_ARM_MAG value to 0.20 from 0.15 seems to eliminate the problem but then "EKF high imu accel bias" pops up instead. Rising the COM_ARM_EKF_AB from 0.0024 to 0.0030 eliminates this (0.0028 also works but not consistently). 

Whether these changes are within a reasonable tolerance level or not is beyond my knowledge so any comments will be appreciated. Might also be that the controller is of low quality.

Both the preflight errors mentioned above does not stop the quadro from arming.

david-cutting, 2018-07-19:
One thing you may want to check... on my Flamewheel clone I didn't have the arrows on the GPS unit and the PX4 lined up. Once I lined them up, things went much smoother.

closed 2018-08-10 by santiago3dr