PX4/PX4-Autopilot #8482: Preflight error msg are spammed in qgc

CarlOlsson, 2017-12-17:
After a flight with master e8624f8afe6199fa8dc9ee1bfc74ac8f6ef52117 QGC started complaining about preflight failure. However the messages were coming once a second and it never stopped until I rebooted, feel free to close this issue if this is intended behavior.
```
ulog_messages log001.ulg 
0:00:06 WARNING: [commander] REJECT POSITION CONTROL
0:04:47 WARNING: [commander] REJECT POSITION CONTROL
0:04:53 WARNING: [commander] REJECT POSITION CONTROL
0:04:59 WARNING: [commander] REJECT POSITION CONTROL
0:05:01 WARNING: [commander] REJECT POSITION CONTROL
0:05:06 WARNING: [commander] REJECT POSITION CONTROL
0:05:14 WARNING: [commander] REJECT POSITION CONTROL
0:11:57 WARNING: [lib__ecl] EKF GPS fusion timeout - reset to GPS
0:12:18 WARNING: [lib__ecl] EKF GPS fusion timeout - reset to GPS
0:12:24 WARNING: [lib__ecl] EKF baro hgt timeout - reset to baro
0:12:24 WARNING: [lib__ecl] EKF baro hgt timeout - reset to baro
0:12:24 WARNING: [lib__ecl] EKF baro hgt timeout - reset to baro
0:12:27 WARNING: [lib__ecl] EKF GPS fusion timeout - reset to GPS
0:12:53 WARNING: [lib__ecl] EKF GPS fusion timeout - reset to GPS
0:13:02 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:02 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:02 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:03 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:04 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:04 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:04 WARNING: [lib__ecl] EKF GPS fusion timeout - reset to GPS
0:13:05 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:06 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:07 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:08 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:09 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:10 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:11 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:12 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:13 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:14 WARNING: [lib__ecl] EKF GPS fusion timeout - reset to GPS
0:13:14 WARNING: [commander_tests] PREFLIGHT FAIL: EKF YAW ERROR
0:13:15 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:16 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:17 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:18 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:19 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:20 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:21 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:22 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
0:13:23 WARNING: [commander_tests] PREFLIGHT FAIL: EKF VEL ERROR
```
log is [here](https://logs.px4.io/plot_app?log=19d124ff-68a3-4f6b-8f16-c542fedc2b9b)

LorenzMeier, 2017-12-17:
The log is certainly abnormal. What immediately raises an eyebrow is that your jamming indicator is all over the place: https://logs.px4.io/plot_app?log=19d124ff-68a3-4f6b-8f16-c542fedc2b9b#Nav-GPS-Noise-_-Jamming

The RAW GPS plot also looks suspicious. I suspect you have a huge EMI issue on your vehicle, potentially due to a low-cost FPV camera?

In the current hardware configuration this is not flyable.

CarlOlsson, 2017-12-17:
Thanks for pointing it out!
No FPV camera yet but the wifi module. Don't remember having so much problems with it before but the gps jamming indicator goes to normal values if I remove it, and also all the other issues with heading innovations etc seems to be resolved.
[image]

However, another thing. After I calibrated all sensors, (with a long usb cable since I removed the wifi) I could not arm because of the failing preflight checks
```
0:00:26 WARNING: [commander_tests] PREFLIGHT FAIL: MAG SENSORS INCONSISTENT
0:00:26 WARNING: [commander] NOT ARMING: Preflight checks failed
0:00:32 WARNING: [commander] NOT ARMING: Preflight checks failed
```
Since I just calibrated the mag I thought I would slightly increase the test limit by increasing the parameter [COM_ARM_MAG](https://github.com/PX4/Firmware/blob/e8624f8afe6199fa8dc9ee1bfc74ac8f6ef52117/src/modules/commander/PreflightCheck.cpp#L231-L231). But this one is no where to be found in QGC, and it is also not visible in the logfile
```
ulog_params sess011/log001.ulg | grep COM_ARM
COM_ARM_AUTH,256010
COM_ARM_EKF_AB,0.00240000011399
COM_ARM_EKF_GB,0.000869999988936
COM_ARM_EKF_HGT,1.0
COM_ARM_EKF_POS,0.5
COM_ARM_EKF_VEL,0.5
COM_ARM_EKF_YAW,0.5
COM_ARM_IMU_ACC,0.699999988079
COM_ARM_IMU_GYR,0.25
COM_ARM_MIS_REQ,0
COM_ARM_SWISBTN,0
COM_ARM_WO_GPS,1
```

closed 2018-08-10 by santiago3dr