PX4/PX4-Autopilot #8508: Throttle stop after front transition

JohnSnowball, 2017-12-21:
**Bug Report** 

Hello,  here's the log link. 
https://logs.px4.io/plot_app?log=f10ed9d8-4eab-4b31-962f-bd5c69178878

After transition to fw is done, the pusher throttle drops to 0 and goes back to normal, then altitude drops as throttle is zero. Check this:
[image]

I reproduced this EVERYTIME I make a front transition. Tried to modify trasition parameters but seems no use.

And I found in source code this, which might be related to the bug:
[image]
I can't fully understand what this set_idle_fw() for, could you add a few more lines of comments here?

dagar, 2017-12-21:
I believe this is largely how TECS is initialized after a VTOL transition.

[image]

Later I'd like to look at having TECS in control of the throttle during transition to avoid the issue.

priseborough, 2017-12-21:
I would add an else to the TECS::_initialize_states function that to handle the case where of an in-air start that is not climbout mode. This would set the throttle integrator to cruise throttle.

dagar, 2017-12-21:
Thanks Paul. ECL PR https://github.com/PX4/ecl/pull/372

JohnSnowball, 2017-12-26:
I made a test for changes from @priseborough  https://github.com/PX4/ecl/pull/372.
The height drop problem is not that bad, however, it still exist. Here is the log:
https://logs.px4.io/plot_app?log=79bd4404-ea40-45f1-be64-99cc9de12835

So I would like to make tecs start in advance, like below, but I'm not sure if this would cause other problems, maybe you could give me some advice. @dagar 
[image]

priseborough, 2017-12-26:
That will cause problems with integrator windup unless there is a method to clamp integrator states to initial values when TECS is not in control. 

TECS demands a nose down pitch transient when it starts. What specific energy and energy rate errors are driving that demand? Are they caused by internal derivative filter switch-on transients or by a mismatch between TECS demands and initial flight condition?

For fixed wing operation TECS has a climbout mode that gives the internal filter time to settle. It is likely an equivalent mode for VTOL launches is required.

JohnSnowball, 2017-12-27:
Thanks for your advice @priseborough, but I don't understand why TECS needs a nose down pitch transient?
When we start TECS just at the moment the front transition is done, all states are changed from 0 to real value, things are discontinue, if we start TECS when vehicle reaches blend speed, I suppose this would allow a few seconds for TECS to follow changes, thus less discontinue. 
And I checked my log above, after front transition is done, the heigt drops due to discontinue, then pitch integrator raises rapidly to compensate that height drop. 
[image]

I think it's discontinue that will accelerate integrator windup, start TECS in advance will ease that point.
Please correct me if I'm wrong, your advice will help me understand better.

priseborough, 2017-12-27:
TECS can be started before being in control to allow internal rate of change filters to settle, but integrator states should be held at initial values when this is being done.

dagar, 2018-02-07:
This PR might help. https://github.com/PX4/Firmware/pull/8835

closed 2018-08-10 by santiago3dr