PX4/PX4-Autopilot #8509: Double cameras with camera_trigger

micheleportioli, 2017-12-21:
Hello everyone,

one problem that @deon and I had is about triggering two cameras independently using camera_trigger driver.
Both AUX5 and AUX6 outputs are activated when I run camera_trigger, but I'd like to control separately the two ports in order to have one camera triggering at at certain time from AUX5 and an other one from AUX6.

Is it possible to restructure the code in order to make it happens? Any suggestions about how to do that?

Thanks!

dagar, 2017-12-21:
Can we go over some of the desired use cases to think through how it should be generalized?

Would anyone realistically want more than 2 cameras?
Should one "camera trigger configuration" have multiple outputs or do we keep it simple and allow 1-X camera_trigger instances, each with a corresponding set of parameters.

mhkabir, 2017-12-22:
It's much better to keep 1-X instances. I tried the first approach sometime back, and it's just not flexible enough.

The only thing about multiple instances is actually dealing with how to handle parameters

dagar, 2017-12-22:
Thanks for the input, and I tend to agree. I've thought about generating param sets for things like this, calibration parameters, multi EKF, etc.

closed 2019-02-19 by stale[bot]