PX4/PX4-Autopilot #8511: Zero offset of SDP3x conflicts with uncalibrated airspeed status

svpcom, 2017-12-21:
If airspeed sensor (SDP3x) has zero offset then SENS_DPRES_OFF sets to 0.0 which is
handled as default (uncalibrated) value. For example qgroundcontrol shows airspeed sensor in bad state.  Do we need to set it to some small value (for example 0.001) or select other value for uncalibrated state?

LorenzMeier, 2017-12-21:
Short-term setting it to a very small value would fix this. Does it really indicate you 0 when you calibrate it? I find that the offset is virtually zero, but still can be calibrated to be completely zero.

svpcom, 2017-12-21:
Yes, in my case it completely zero.

dagar, 2017-12-23:
Interesting problem to have!

I think what we really need is a proper interface for sensor status to send QGC. Right now it directly checks these parameters. @DonLakeFlyer FYI

LorenzMeier, 2018-01-01:
Fixed here: https://github.com/PX4/Firmware/commit/ddf0ecfc3801b63f0849d5ba37a523746c53d8b4

closed 2018-01-01 by LorenzMeier