PX4/PX4-Autopilot #8512: Simplify multi vehicle setup

LorenzMeier, 2017-12-21:
Here is a multi-vehicle scripting system:

https://github.com/acsl-mipt/multiple-sitl/issues/3

@korotkoves Would you be willing to send a pull request against PX4/Firmware to include your script subsystem in a folder?

Current ROS docs:
https://dev.px4.io/en/simulation/multi-vehicle-simulation.html

Docs issue:
https://github.com/PX4/Devguide/issues/384

korotkoves, 2017-12-22:
Ok, I can start to code in a week, may be later. It needs pull request to PX4/sitl_gazebo either.

mzahana, 2017-12-22:
I wanted to do something more generic similar to this wih a Qt GUI. Good to see some work on this.

korotkoves, 2018-01-16:
I generated pull request to sitl_gazebo (https://github.com/PX4/sitl_gazebo/pull/166) and new branch in multiple-sitl project (https://github.com/acsl-mipt/multiple-sitl/tree/1.7)
To test simply clone multiple-sitl into Firmware/Tools and replace sitl_gazebo with https://github.com/acsl-mipt/sitl_gazebo/tree/1.7 in Firmware/Tools.
If proposition needs to be in smth different way, please tell me.

weiminshen99, 2018-03-04:
Hi, @korotkoves,
Thanks for your contributions and I am very interested in using your code to run multi-sitl without ros. 
I have cloned your directories into my Firmware/Tools based on your post on Jan 16 above, could you please tell me how to run your code or where I can find instructions for that? Thanks!

korotkoves, 2018-03-06:
@weiminshen99, to run without ROS
install dependencies and build modules:
sudo ./install/sudo_deps.sh
./install/prepare.sh

run:
./start.rb --nomavros -n NUM
 where NUM - number of copters
./start.rb -h
for options list

weiminshen99, 2018-03-06:
Thank you so much. I will give it try and let you know the results. 

Wingman

weiminshen99, 2018-03-08:
Hi, korotkoves,

Thanks again for your help and I can now  create multiple px4/iris in gazebo without mavros. 

However, I cannot connect to them using their ports listed. For example, my offboard  program cannot connect to iris1 on port 15016, nor to other px4/iris using their corresponding ports. 

Would you please tell me how to proceed from here? Thanks. 

Wingman

korotkoves, 2018-03-08:
According to README ports are 15011, 15021, ...
To debug ./start.rb --debug --nomavros -n NUM

weiminshen99, 2018-03-18:
Hi, Korotkoves,

Thanks for your patience. Here is my test results:

	Without mavros, I tried all four ports 15010, 15011, 15015, 15016, but cannot connect to any of these ports.
	With mavros, I can connect to 15016, but the UAV does not response to any command that has sent to it.

Please help. Do you have any testing programs that I can try? Thanks!

Wingman

korotkoves, 2018-03-20:
Use QGroundControl to connect to 15010 or 15011 port
If no connect, see debug xterm console of px4, may be some error in it (on px4 with version 1.7.3 it worked)
Reed main documentation of PX4 project.

weiminshen99, 2018-03-23:
Dear Korotkoves,

I want to share the good news with you that your branch 1.7 works very nicely for me. I can now run without mavros three PX4s using different offboard controllers and companion computers. Thank you very much for your help and I greatly appreciate that.

In the future, if I would like to keep using your system with the latest PX4 (your branch 1.7 uses PX4 version 1.7.3), is there an easy way to set that up? Please let me know.

Have a great day!

Wingman

weiminshen99, 2018-03-24:
Dear @korotkoves,

I like your branch 1.7 very much, and I am now trying your branch "for-px4-pr" as instructed:
    1. Clone https://github.com/acsl-mipt/multiple-sitl/tree/for-px4-pr into Firmware/Tools/  (branch v1.7.3)
    2. Replace sitl_gazebo with https://github.com/acsl-mipt/sitl_gazebo/tree/1.7 in Firmware/Tools 
Then, I cd into Firmware/Tools/multiple-sitl/install/, and did
    3. sudo ./sodu_deps.sh
    4. ./prepare.sh
After that, I cd into .., and did
    5. ./start.rb --nomavros --debug -n 3
This created 3 px4s and gazebo, but the px4s did not connect to gazebo and just wait there.

Could you please tell me what went wrong? More importantly, is it possible to do step 5 without step 2? That way, we can use the existing sitl_gazebo in Firmware/Tools, and thus keep up with the future versions of Firmware. Thank you!

korotkoves, 2018-03-25:
I checked steps from 1 to 5 on my computer with Ubuntu 16.04 again, and it works as expected.
For example 3 instance of px4 says:
INFO  [simulator] Waiting for initial data on UDP port 15039. Please start the flight simulator to proceed..
INFO  [simulator] Got initial simulation data, running sim..
...
and gazebo says:
[Dbg] [common.h:74] get mavlink_udp_port 15039 for iris3 model from default.xml
...

Step 5 is not possible without step 2, because multiple-sitl uses modifications of sitl_gazebo.

As I understand this issue is exactly for integrating multiple-sitl and sitl_gazebo modification to main PX4 tree.

korotkoves, 2018-04-12:
Branch https://github.com/acsl-mipt/multiple-sitl/tree/for-px4-pr was simplified and redone for https://github.com/acsl-mipt/sitl_gazebo/tree/multiple-sitl-pr (https://github.com/PX4/sitl_gazebo/pull/199)

closed 2018-08-10 by santiago3dr