PX4/PX4-Autopilot #8514: Out of flash room?

dposea, 2017-12-21:
**Bug Report** 

I just downloaded the source and tool chains to Ubuntu. I got the following error:

ninja: build stopped: subcommand failed.
Makefile:153: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1
dgp@dgp-M18xR1:~/src/Firmware$ make V=1 px4fmu-v2_default
[1/3] Linking CXX executable nuttx_px4fmu-v2_default.elf
FAILED: : && /usr/lib/ccache/arm-none-eabi-g++   -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wno-overloaded-virtual -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG  -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -nodefaultlibs -nostdlib -Wl,--warn-common,--gc-sections src/firmware/nuttx/CMakeFiles/nuttx_px4fmu-v2_default.elf.dir/__/__/platforms/empty.c.obj  -o nuttx_px4fmu-v2_default.elf -L/home/dgp/src/Firmware/build/px4fmu-v2_default/external/Install/lib -T/home/dgp/src/Firmware/build/px4fmu-v2_default/NuttX/nuttx/configs/px4fmu-v2/scripts/ld.script -Wl,-Map=/home/dgp/src/Firmware/build/px4fmu-v2_default/nuttx_px4fmu-v2_default.map -Wl,--warn-common -Wl,--gc-sections -Wl,--start-group msg/libuorb_msgs.a src/drivers/airspeed/libdrivers__airspeed.a src/drivers/boards/common/libboards_common.a src/drivers/boards/common/stm32/libboard_stm32.a src/drivers/boards/px4fmu-v2/libdrivers__boards__px4fmu-v2.a src/drivers/camera_trigger/libdrivers__camera_trigger.a src/drivers/device/libdrivers__device.a src/drivers/frsky_telemetry/libdrivers__frsky_telemetry.a src/drivers/gps/libdrivers__gps.a src/drivers/hmc5883/libdrivers__hmc5883.a src/drivers/l3gd20/libdrivers__l3gd20.a src/drivers/led/libdrivers__led.a src/drivers/lis3mdl/libdrivers__lis3mdl.a src/drivers/ll40ls/libdrivers__ll40ls.a src/drivers/lsm303d/libdrivers__lsm303d.a src/drivers/mpu6000/libdrivers__mpu6000.a src/drivers/mpu9250/libdrivers__mpu9250.a src/drivers/ms4525_airspeed/libdrivers__ms4525_airspeed.a src/drivers/ms5525_airspeed/libdrivers__ms5525_airspeed.a src/drivers/ms5611/libdrivers__ms5611.a src/drivers/pwm_input/libdrivers__pwm_input.a src/drivers/pwm_out_sim/libdrivers__pwm_out_sim.a src/drivers/px4flow/libdrivers__px4flow.a src/drivers/px4fmu/libdrivers__px4fmu.a src/drivers/px4io/libdrivers__px4io.a src/drivers/rgbled/libdrivers__rgbled.a src/drivers/sdp3x_airspeed/libdrivers__sdp3x_airspeed.a src/drivers/sf0x/libdrivers__sf0x.a src/drivers/sf1xx/libdrivers__sf1xx.a src/drivers/stm32/libdrivers__stm32.a src/drivers/stm32/adc/libdrivers__stm32__adc.a src/drivers/stm32/tone_alarm/libdrivers__stm32__tone_alarm.a src/drivers/teraranger/libdrivers__teraranger.a src/drivers/vmount/libdrivers__vmount.a src/modules/sensors/libmodules__sensors.a src/systemcmds/bl_update/libsystemcmds__bl_update.a src/systemcmds/hardfault_log/libsystemcmds__hardfault_log.a src/systemcmds/mixer/libsystemcmds__mixer.a src/systemcmds/mtd/libsystemcmds__mtd.a src/systemcmds/param/libsystemcmds__param.a src/systemcmds/perf/libsystemcmds__perf.a src/systemcmds/pwm/libsystemcmds__pwm.a src/systemcmds/reboot/libsystemcmds__reboot.a src/systemcmds/top/libsystemcmds__top.a src/systemcmds/ver/libsystemcmds__ver.a src/modules/camera_feedback/libmodules__camera_feedback.a src/modules/commander/libmodules__commander.a src/modules/events/libmodules__events.a src/modules/land_detector/libmodules__land_detector.a src/modules/load_mon/libmodules__load_mon.a src/modules/mavlink/libmodules__mavlink.a src/modules/navigator/libmodules__navigator.a src/modules/ekf2/libmodules__ekf2.a src/modules/fw_att_control/libmodules__fw_att_control.a src/modules/fw_pos_control_l1/libmodules__fw_pos_control_l1.a src/modules/gnd_att_control/libmodules__gnd_att_control.a src/modules/gnd_pos_control/libmodules__gnd_pos_control.a src/modules/mc_att_control/libmodules__mc_att_control.a src/modules/mc_pos_control/libmodules__mc_pos_control.a src/modules/vtol_att_control/libmodules__vtol_att_control.a src/modules/logger/libmodules__logger.a src/modules/dataman/libmodules__dataman.a src/modules/systemlib/libmodules__systemlib.a src/modules/systemlib/param/libmodules__systemlib__param.a src/modules/uORB/libmodules__uORB.a src/lib/controllib/liblib__controllib.a src/lib/conversion/liblib__conversion.a src/lib/DriverFramework/framework/liblib__DriverFramework__framework.a src/lib/ecl/liblib__ecl.a src/lib/geo/liblib__geo.a src/lib/geo_lookup/liblib__geo_lookup.a src/lib/launchdetection/liblib__launchdetection.a src/lib/led/liblib__led.a src/lib/mathlib/liblib__mathlib.a src/lib/mathlib/math/filter/liblib__mathlib__math__filter.a src/lib/mixer/liblib__mixer.a src/lib/runway_takeoff/liblib__runway_takeoff.a src/lib/tailsitter_recovery/liblib__tailsitter_recovery.a src/lib/version/liblib__version.a src/platforms/common/libplatforms__common.a src/platforms/nuttx/libplatforms__nuttx.a src/platforms/nuttx/px4_layer/libplatforms__nuttx__px4_layer.a NuttX/apps/libapps.a NuttX/nuttx/arch/arm/src/libarch.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/libc/libc.a NuttX/nuttx/configs/libconfigs.a NuttX/nuttx/libxx/libcxx.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -Wl,--end-group -lm -lgcc ROMFS/libromfs.a src/lib/DriverFramework/framework/src/libdf_driver_framework.a && :
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: nuttx_px4fmu-v2_default.elf section `.data' will not fit in region `flash'
/usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: region `flash' overflowed by 2377 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Makefile:153: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1
dgp@dgp-M18xR1:~/src/Firmware$ make pix4fmu-v2_default
Makefile:360: *** "pix4fmu-v2_default cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop.

dposea, 2017-12-21:
I found similar issues, commented out a couple of modules,and it's fine. I'm leaving this open because the default build is broken.

dagar, 2017-12-22:
Update GCC to 5.4 (used for the official builds: http://ci.px4.io:8080/blue/organizations/jenkins/Firmware/detail/master/137/pipeline/38) or newer. Currently px4fmu-v2_default is very tight on available flash, but it's fine with more recent GCC. https://dev.px4.io/en/setup/dev_env.html

closed 2017-12-22 by dagar

dposea, 2017-12-22:
I am using 5.4. Wait, maybe the EABI toolchain is older. I got a copy from the repo, but that script ubuntu_sim.sh may grab another version.
I'll check to be sure and you can close the issue.
Thanks 

Closed #8514.

dposea, 2017-12-22:
That was what happened. Arm-none-eabi was 4.9.

 
  On Thu, Dec 21, 2017 at 19:40, David Posea<dposea@yahoo.com> wrote:   I am using 5.4. Wait, maybe the EABI toolchain is older. I got a copy from the repo, but that script ubuntu_sim.sh may grab another version.
I'll check to be sure and you can close the issue.
Thanks