PX4/PX4-Autopilot #8515: Go to DESCEND mode while GPS always connected

JohnSnowball, 2017-12-22:
**Bug Report** 

I got a crash during a vtol flight, the flight mode changed to "NAVIGATION_STATE_DESCEND" at the end of front transition. Then it's goes lower and lower, when I found it is wrong, I tried to save it by turn on RTL, but it's too late.

I checked source code, it seems the only way to go into that mode is GPS timeout, but I checked my GPS, at least the altitude data looks good.
here's my log link:
https://logs.px4.io/plot_app?log=d5bbeb04-40dc-464b-bcd8-408cad3983f5

dagar, 2017-12-22:
This was an estimator problem.

[image]

[image]

[image]

According to estimator_status.filter_fault_flags it was bad accelerometer data.

JohnSnowball, 2017-12-22:
@dagar So first is accelerometer fault -> ekf estimator got a wrong vel/pos estimation -> afterwards although GPS works well, ekf2 rejects GPS fusion as there's a big difference -> ekf2 thinks GPS timeout.

Am I correct?

JohnSnowball, 2017-12-22:
[image]
And I saw this abnormal sampling regularity, what does it mean? Does it mean my accelerometer has something wrong?

dagar, 2017-12-22:
I think those are just missed samples in the log, but I'd have to look at what flight review is doing. If you have a higher quality sdcard it usually helps. I would guess your problems are vibration, but it's hard to see in the log due to filtering. That's something we're hoping to improve in the near future.

Does that line up with what you see first hand? See if you can tighten things, balance props, etc, and if it's still an issue we can adjust the EKF.

dagar, 2017-12-22:
You can see it corresponding with the pusher.

[image]

dagar, 2017-12-22:
Once more with the solution status flags as well (scaled to fit).

[image]

dagar, 2017-12-22:
Closing as this isn't really a Firmware issue, but I still hope we can significantly improve the detection and handling of these issues (with updated IMU drivers and different filtering) for the next PX4 release.

closed 2017-12-22 by dagar

Antiheavy, 2017-12-22:
> If you have a higher quality sdcard it usually helps.

We have started using the fastest available U3 rated SDcards we can find in all our PX4 based autopilots.  "Normal" SDcards are too slow and always have many small gaps in the data logs.

> I would guess your problems are vibration, but it's hard to see in the log due to filtering. That's something we're hoping to improve in the near future.

+1 for improved filtering (turning off internal IMU filters) so that vibration can properly be measured/detected!

> So first is accelerometer fault -> ekf estimator got a wrong vel/pos estimation -> afterwards although GPS works well, ekf2 rejects GPS fusion as there's a big difference -> ekf2 thinks GPS timeout.

This happens to us sometimes too.  I hope there will be some improvements coming to ekf2 to make it more robust eventually...

JohnSnowball, 2017-12-23:
@dagar  I checked other log files, I couldn't found bad raw acceleration data like this during transition.
I agree that it's not firmware bug, it could be some mechanical problem. Next time I will do more pre-checks.
@Antiheavy I prefer to make more checks on my side, we can't throw everything to EKF2. I wish both of us have less chance of crash. XD