PX4/PX4-Autopilot #8518: ekf2 failed to set home when ekf2_aid_mask is setup as gps+vpe position(9)

EliaTarasov, 2017-12-22:
PX4 version: master
When EKF2_AID_MASK is setup as fuse GPS and Vision Position at the same time, then home pos is failed to be setup. However adding vision position fusion in flight works ok. Is it possible to setup home pos using GPS and Vision Position at the same time?

dagar, 2017-12-22:
Do you have a log showing this?

EliaTarasov, 2017-12-23:
Yes sure.
AID_MASK 1: https://logs.px4.io/plot_app?log=0637d591-63a8-4354-a4d6-cfd4510392ce
AID_MASK 9: https://logs.px4.io/plot_app?log=56b14d7e-2518-4f55-b7c3-432e385b50cd
Thanks in advance!

ChristophTobler, 2017-12-27:
@EliaTarasov What commit are you using? I just tried with latest master and it works in SITL (I saw you tested in SITL as well).

Can you write down instructions to reproduce it?

ChristophTobler, 2018-01-04:
@EliaTarasov Please reopen and add instructions if this is still an issue.

closed 2018-01-04 by ChristophTobler

EliaTarasov, 2018-01-26:
Confirmed, latest master works ok.