PX4/PX4-Autopilot #8520: Feeding mocap pose as vision pose into EKF2 does not wrok

mzahana, 2017-12-22:
Since mocap position fusion is not supported in EKF2, I tried to feed it as vision position using `vision_pose` plugin in MAVROS. For that, I did the following.
1. I set EKF aid mask to fuse vision position and yaw from vision, and not to fuse GPS
2. I used `vrpn_client_ros` plugin to get mocap data which is published as pose stamped in ENU
3. I relayed the pose to  `/mavros/vision_pose/pose` which is the same type (pose stamped)
4. I see a message on QGC saying something like ' IMU quaternion detected...'
5. I check the `LOCAL_POSITION_NED` in QGC and the position does not match mocap at all
6. The horizon starts to diverge in QGC and does react as expected

If I used LPE and set the params accordingly, I could get mocap data correctly.

So what could be the problem on the EKF2 side? Am I missing more params settings, or this method does not work in the first place?
I am using PX4 v1.7, Ubuntu 16 on Odroid XU4,  mavros Kinetic version

Thanks.

mhkabir, 2017-12-22:
Can you send a log with logging from boot? (use the SDLOG_MODE parameter). The log should have vision data stream started sometime after boot so that everything is captured.

mzahana, 2017-12-24:
Hi I have some updates. I got the mocap fused relatively fine with EKF2. I could do position hold indoors. But it seems that the vision estimates have not fast enough convergence. Compared to LPE, when in position hold mode, I use roll/pitch RC stick to change the drone position. When It is centered, with LPE, it converges very fast to a position once you leave the stick centered. Using EKF2, however, it drifts a bit around  certain position before it converges. It is not very slow convergence but noticeable.

Here is what I did for that to work initially fine.
1. Aligned drone with x-axis of mocap system (OptiTrack)
2. Drone was almost at the centere (0,0) of mocap coordinates
3. Drone was powered while it's still aligned with x-axis of mocap (before I used to power it on in arbitrary heading and position w.r.t. mocap coordinates, and was not giving meaningful pose (local_ned)!)
4. on the companion computer, odroid xu4, I used `vrpn_client_ros` package to get mocap data and then relayed it to `mavros/vision_pose/pose` topic.

With that, all worked fine. However, as I mentioned, the convergence, while in position hold mode is noticeably slower compared to LPE.

Also, there was an issue happened during the position hold mode. It suddenly failed safe to altitude hold mode, and reported that local position was lost. Then I re-switched back to position hold mode and it worked.

A log of the flight is in the following link.
https://logs.px4.io/plot_app?log=1addab1f-3044-4aba-91b2-0401477717bf

Thanks.

mzahana, 2017-12-24:
Here is my paramters setting in QGC
[image]

[image]

mzahana, 2017-12-24:
Another concern is, the (0,0,0) point of the local position that is created by autopilot is with respect to the point where you started to get vision data, and it does not necessarily match the vision data. But it is more convienent when working with mocap to have local position always matching the mocap coordinates, just like what happens in LPE. The only way to do this with vision estimates fusion in EKF2 is to start feeding vision when the drone at the origin of mocap. Assuming that you work with multiple drones, then all of them have to be initialized at mocap origin, which is different form LPE where you can initialize the drone any where in your captured space.

Any way to go about this in EKF2 other than the method I mentioned?

Thanks.

LorenzMeier, 2017-12-24:
@priseborough Could you have a look at the log and see what we can do to improve this?

What is important to note is that mocap is != vision - it is more of a GPS replacement including heading. So the vision fusion is probably not the right approach - we rather should have a separate data path that fuses mocap like GPS and also uses the heading as a magnetometer replacement.

We also certainly need an automatic calibration step where the drone calibrates to the transformation from its own Frame to MOCAP. Not sure what to do best there (reset or store an offset). Starting with a reset to the frame might be best.

mzahana, 2017-12-24:
@LorenzMeier  agree on the separate data paths to include MOCAP in EKF2 pipeline.

I found a related attempt in the following issue.
https://github.com/PX4/ecl/issues/271

priseborough, 2017-12-26:
Can you please provide your parameters. For some reason many parameters including those for EKF2  are missing from the log.

priseborough, 2017-12-29:
@mzahana Can you also please provide a log with pre-arm logging enabled so that it can used with replay to test any changes.

mzahana, 2017-12-29:
@priseborough Do you want the parameters for the log attached in this conversation, or a new one for the same attempt (using mocap as vision), but log starting from boot?

I am not sure If I have the same parameters for the log I attached earlier; there have been many changes.

priseborough, 2017-12-29:
The new log please

mzahana, 2017-12-30:
@priseborough here is a new log from a new flight.
https://review.px4.io/plot_app?log=f324350a-c70b-44cb-978e-4b27f10280a1

* mocap used is OptiTrack 
* in motive, in the streaming window, I chose that z-axis to be considered up direction instead of the default y-axis 
* aligned vehicle forward direction with x-axis of mocap 
* defined rigidbody 
* used onboard computer which runs ROS, and connected to the flight controller (MindPX)
* used `vrpn_client_ros` package to get pose data form mocap to ROS 
* used `topic_tools` package to relay the mocap pose topic to `/mavros/vision_pose/pose` topic 
* in QGC, chose `EKF2` as estimator 
* modified `EKF2_AID_MASK` paramter to disable GPS and enable vision fusion as well as yaw fusion from vision 
* did flight in altitude/posigion hold modes 

In position hold mode, when traveling to new position then releasing the RC sticks, it takes a noticeable time to converge to the position. When using LPE, this is not noticeable.

Here is the parameter file of the flight

[indoor_ekf2_mocap.params.zip](https://github.com/PX4/Firmware/files/1594296/indoor_ekf2_mocap.params.zip)

Thanks!

mhkabir, 2017-12-30:
Hmm can you try reducing the `VIS_DELAY` parameter to zero? If the timestamps are synchronised with mavros and `vrpn_client_ros` gives correct timestamps, the parameter should be zero.

mzahana, 2017-12-30:
@mhkabir How to check that
1. timestamps are synchronized with mavros
2. `vrpn_client_ros` is giving correct timestamps? it's publishing changing time stamps, but I did not confirm if they are correct.

I just would like to understand how time stamping and synchronization affect the fusion here. Thanks.

priseborough, 2017-12-30:
@mzahana The ekf2 runs the EKF prediction and correction processing on a delayed time horizon and a predictor algorithm using IMU data only is used to propagate the EKF states forward to the current time horizon. Observation data is put into buffers along with the time stamp, and is only retrieved from the buffer and fused when the time stamp falls behind the EKF's delayed time horizon.

I notice in your data that the innovations rise significantly whenever the vehicle accelerations which is consistent with the position observations being fused at the wrong time.

mzahana, 2017-12-30:
Thanks for the notes.
I am interested to understand this a bit more so I can better analyze the logs myself in current future situations.
I know from KF theory that 
1. a process model is used to predict the states from current time `x_pred(t)` to next time `x_pred(t+1)`
2. Then, a measurement model is used to do the correction step using observations, `y(t)`. So, you get `x_corrected(t+1)`

What is unclear to me, is why the EKF is implemented on a delayed time horizon?

And, yes, It makes sense to me to use data that corresponds to EKF in time and not future ones. But still can not see how this is related to EKF implementation on a delayed horizon.

Thanks!

priseborough, 2017-12-30:
All sources of data arrive with different time delays so we either:

a) Perform correction steps when data arrives, and increase observation noise to allow for increased error when manoeuvring.
b) Perform correction steps using state vector from an earlier time corresponding to the measurement time and covariance matrix from current time. This can destabilise the filter and introduces errors due to the covariance matrix being inconsistent.
c) Run the prediction and correction on a delayed time horizon so that the observations, states and covariances are all fro the same point in time. This is the method used by ekf2.

mzahana, 2017-12-30:
Thanks. Is there some kind of documentation/paper that explains this approach?

mhkabir, 2017-12-31:
@mzahana As Paul says, the innovation rise with body accelerations indicates that the timestamping is off.

> 1. timestamps are synchronized with mavros
2. vrpn_client_ros is giving correct timestamps? it's publishing changing time stamps, but I did not confirm if they are correct.

1. Should be done automatically. 
2. This is harder. I'm not sure if the package actually syncs time with the computer running Motive. You would need to check the source code and trace how it gets the timestamp. It should *not* just be `ros::Time::now()`. If it is, then you will need to set/tune the EKF2 vision delay parameter appropriately to compensate.

mzahana, 2017-12-31:
@mhkabir I checked the source code of `vpn_ros_client` package and they report two types of timestamps according to a parameter. See [code lines here](https://github.com/ros-drivers/vrpn_client_ros/blob/kinetic-devel/src/vrpn_client_ros.cpp#L169-L177)
1. It either gets the time reported from the mocap side or,
2. uses `ros::Time::now()`

For 2nd case what is rule for setting the delay for the EKF2 vision delay parameter?

Thanks.

mzahana, 2017-12-31:
@priseborough which log variables show the rise in innovation?

priseborough, 2017-12-31:
[image]

mzahana, 2018-01-02:
Achieved better flight (and EKF  estimation ) after reducing the `EKF2_EV_DELAY` to `50.0 ms` instead of the default value `175.0 ms`.

Log is [here](https://logs.px4.io/plot_app?log=a6ca73bc-7498-4178-bd1f-23bbfe78ec0b)

@priseborough  So, what's the plan for mocap fusion in EKF2? Is it a separate data path, or through vision topics?

Thanks.

priseborough, 2018-01-02:
There is enough flexibility in the ecl EKF external vision interface to handle this use case, so I will now look at  what modifications are required to handle initialisation of the position so that the origin is the same as that used by the mocap.

priseborough, 2018-01-02:
Something else you can do to improve accuracy further is to set EKF2_IMU_POS_X,Y,Z and EKF2_EV_POS_X,Y,Z to the position of the IMU sensor and mocap reference point. Setting them to the nearest 0.01m (1 cm) should be good enough.

mzahana, 2018-01-02:
@priseborough Thanks!

> can set EKF2_IMU_POS_X,Y,Z and EKF2_EV_POS_X,Y,Z to the position of the IMU sensor and mocap reference point.

Those parameters are with respect to the center of the flight controller ? If that's the case, I guess I can set rough estimates of the `EKF2_IMU_POS_X`/Y/Z, with X along the forward direction, Y along right direction, Z down w.r.t FCU.

For `EKF2_EV_POS_X`/Y/Z, what do you mean by the mocap reference point? The mocap origin is always (0,0,0)

mhkabir, 2018-01-02:
>  what do you mean by the mocap reference point?

The point on the vehicle defined as the origin of the rigid body.

mzahana, 2018-01-02:
@mhkabir OK. So you mean the origin point (of the set of markers that define the rigid body) w.r.t to the FCU center?

priseborough, 2018-01-02:
@mzahana The parameters are relative to whatever point on the vehicle you wish to use as the reference for navigation and control. For multi-rotors this is best kept close to the c.g to avoid coupling between angular rates and position that can reduce stability of velocity control loops.

By  'mocap reference point' I mean the point on the body where mocap system places the body frame origin.

mzahana, 2018-01-04:
@priseborough understood. Thanks.

priseborough, 2018-01-07:
@mzahana I have checked through the code and in the absence of GPS aiding (make sure you disable GPS fusion using EKF2_AID_MASK) it should reset the position to the initial mocap position when starting.

mhkabir, 2018-04-04:
Closing as stale. Should be working now.

closed 2018-04-04 by mhkabir