PX4/PX4-Autopilot #8522: V1.7.2 HITL failed

ZingHD, 2017-12-23:
**Bug Report** 
The firmware upgrade to v1.7.2, the hardware in the loop simulation, the sensor will prompt detection failed, the simulation will be flight fault
[image]

dagar, 2017-12-23:
What's your setup? Is this jmavsim? When was it last working?

TSC21, 2017-12-24:
@arwagoner probably you are facing this same problem with Gazebo ATM, which means the problem may be coming from the Firmware side. Can you expose here your findings while integrating the HIL functionality for Gazebo? Thanks

ZingHD, 2017-12-25:
@dagar  The firmware 1.6.5 HITL is work，I can easily debug parameters。

LorenzMeier, 2017-12-27:
@ZingHD I've fixed this here: https://github.com/PX4/Firmware/pull/8532

closed 2017-12-27 by LorenzMeier

ZingHD, 2017-12-28:
@LorenzMeier  I had change the three files but it not fixed
src/modules/commander/state_machine_helper.cpp
src/modules/sensors/voted_sensors_update.cpp
src/modules/commander/commander.cpp

LorenzMeier, 2017-12-28:
It definitely is. If it didn't work for you then either the changes were not consistent or you didn't configure the system correctly.

LorenzMeier, 2017-12-28:
Also note the documentation here:
https://dev.px4.io/en/simulation/hitl.html

LorenzMeier, 2017-12-28:
I've found one caveat - if you launch QGC after jMAVSim the initial launch can eat all of the CPU and the simulation diverges.

arwagoner, 2017-12-28:
These changes worked for HIL in Gazebo. Thanks! 
[image]

ZingHD, 2017-12-29:
@LorenzMeier  It works on Gazebo SITL thank you very much. @arwagoner I can only found the document
https://dev.px4.io/en/simulation/gazebo.html use Gazebo SITL,can you tell me how to use Gazebo HITL，I  well use HITL ，I want to test a week to determine the stability of the flight control，thanks。

arwagoner, 2017-12-29:
@ZingHD Gazebo HITL has not been merged yet https://github.com/PX4/sitl_gazebo/pull/146. The docs will be updated shortly with instructions on how to use it.

ZingHD, 2017-12-31:
@arwagoner Thank you answer.