PX4/PX4-Autopilot #8523: Pixhackv3 sensor position issue with 1.7

urlu75, 2017-12-24:
**Bug Report** 

Accell sensor with 1.7 are not correct. To calibrate them correctly, I have to change the orientation of the board : roll 180 and yaw 270. With that the frame position is correctly detected.
But then, compass calibration is not successful.

Everything is perfectly detected and calibrated with 1.6.5 without any change in the board orientation.

dagar, 2017-12-24:
PX4 v1.7.0 added board detection for px4fmu-v2/v3 boards that actually had differences. We need to figure out how the Pixhawkv3 should be handled, and how it's currently detected.

Could you run `ver all` on the console?
Do you have a schematic for the Pixhackv3? I mainly want to know about the sensors.

urlu75, 2017-12-25:
here is the `ver all`

`
NuttShell (NSH)
nsh> ver all
HW arch: PX4FMU_V2
HW type: V2M
HW version: 0x0009000A
HW revision: 0x00000000
FW git-hash: 7dab5d438027c0b43c8b269b403a21e92dec7061
FW version: Release 1.7.2 (17236735)
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: b18053574bf41712cef93e31bf178518f451a350
Build datetime: Dec 21 2017 08:04:32
Build uri: localhost
Toolchain: GNU GCC, 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
MFGUID: 3030383034355113002e0030
MCU: STM32F42x, rev. 3
UID: 2E0030:34355113:30303830 
`

I don't have a schematic for this board, will try to find it.
As it is perfectly working with 1.6.5, would it be possible to guess sensors information ?

dagar, 2017-12-26:
It's being detected as a pixhawk mini (`HW type: V2M`). The px4fmu-v2 version detection is relatively new. We'll likely need to get our hands on a schematic to figure out how to handle this board.

@davids5 FYI

LorenzMeier, 2018-01-01:
@urlu75 What sensors are on this board? What do you get with 1.6.5 here?

[image]

LorenzMeier, 2018-01-01:
Please try this binary which is based on this branch: https://github.com/PX4/Firmware/pull/8553

[px4fmu-v3_default.px4.zip](https://github.com/PX4/Firmware/files/1596218/px4fmu-v3_default.px4.zip)

urlu75, 2018-01-04:
The board is detected as a v2, I tried v2 and v3 FW.

Bellow is the device list with px4fmu-v3_default.px4 1.6.5
[image]

with px4fmu-v3_default.px4 1.7.2
[image]

and with the FW you sent. Seems better, no more warning about gyro but alignement is still not good
[image]

I sent an email to CUAV to request the schematic but no answer yet.

urlu75, 2018-01-23:
If it's help, found what's look like to be the PCB schematic

https://github.com/cuav/CUAV_Hardware/tree/master/Pixhack_V3

LorenzMeier, 2018-05-27:
Is this issue still pending?

dagar, 2018-05-27:
@urlu75 could you quickly verify the internal sensors and orientations on a pixhack v3?

urlu75, 2018-05-27:
yes but ... how do you do that, what would you need ?

davids5, 2018-08-14:
@urlu75 please test with https://github.com/PX4/Firmware/pull/10231 All sensors are now detected. The rotations may need to be adusted.

urlu75, 2018-08-16:
just flash with last dev fw: 3 accel, 3 gyro and 2 mag. seems all good, thanks !
[image]

closed 2018-08-16 by mhkabir