PX4/PX4-Autopilot #8524: Add a new daemon app

KatelynLiu, 2017-12-25:
**Add a new daemon app** 

I want to use a external IMU on pixhawk 2 through TELEM2, so I write a driver as "xsens" and put it under /src/drivers/. However, there is a problem that I can't solve.
If I published the information I handled, and subscribed in the xsens.cpp, I could get the topic. But if I subscribed the topic in another file, there was no message in the topic. And then, I turned the app into a daemon app, but I type "xsens start" by shell in Qgroundcontrol Mavlink Console at once, the communication would be lost in a few second. What's more, this app doesn't auto run either. I don't know why this problem occur. And I wonder how to make my app to be a daemon app and auto run.
I attach my daemon app's main code and CMakeLists.txt below in convenience for the problem to be solved.
Looking forward to your help!

main code:
[https://github.com/KatelynLiu/Firmware/blob/master/src/drivers/xsens/xsens.cpp](url)

CMakeLists.txt:
[https://github.com/KatelynLiu/Firmware/blob/master/src/drivers/xsens/CMakeLists.txt](url)

dagar, 2017-12-26:
You need to add `xsens start` to the init script (https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rcS) to autostart.

Make sure you subscribe in the same thread that you use to copy the data.

KatelynLiu, 2017-12-26:
Thanks for your advice, add `xsens start` do work. However, there's an another problem I mentioned influence the whole project.
If I use `while( !_task_should_exit )`loop in my app to pulish the topic, the pixhawk would discounted at once the app started. And because of this problem, the pixhawk can't connect the QGC at all after I add `xsens start` in the init script. I wonder what's wrong with it. Is it a memory management problem? How can I solve it?

yangang123, 2017-12-27:
@KatelynLiu：
The task is always ready: no delay, or pend????

KatelynLiu, 2017-12-27:
I've solved this problem, thank you all~

closed 2017-12-27 by KatelynLiu

yingzongniubi, 2022-04-28:
> 我已经解决了这个问题，谢谢大家~特别是@dagar和@yangang123

hello! I encountered the same problem as you,i want to control my UAV with ROS2 and use px4_fmu-v5_rtps.it needs type "micrortps_client start(Officially available modules)"by shell in Qgroundcontrol Mavlink Console,then the communication would be lost quicklly.Can you tell me how you solved it?Please forgive my poor English,I would be very grateful if you could reply to me.I am an ordinary college student. : )