PX4/PX4-Autopilot #8525: adis16448 driver question??

yangang123, 2017-12-25:
**issue Report:**

1. 
```
#define ADIS16448_GYRO_DEFAULT_RATE					100
#define ADIS16448_GYRO_DEFAULT_DRIVER_FILTER_FREQ	30

#define ADIS16448_ACCEL_DEFAULT_RATE				100
#define ADIS16448_ACCEL_DEFAULT_DRIVER_FILTER_FREQ	30

#define ADIS16448_MAG_DEFAULT_RATE					100
#define ADIS16448_MAG_DEFAULT_DRIVER_FILTER_FREQ	30

#define ADIS16448_ACCEL_MAX_OUTPUT_RATE              1221
#define ADIS16448_GYRO_MAX_OUTPUT_RATE 1221
```

2. 
```
_gyro_filter_x(ADIS16448_GYRO_DEFAULT_RATE, ADIS16448_GYRO_DEFAULT_DRIVER_FILTER_FREQ),
	_gyro_filter_y(ADIS16448_GYRO_DEFAULT_RATE, ADIS16448_GYRO_DEFAULT_DRIVER_FILTER_FREQ),
	_gyro_filter_z(ADIS16448_GYRO_DEFAULT_RATE, ADIS16448_GYRO_DEFAULT_DRIVER_FILTER_FREQ),
	_accel_filter_x(ADIS16448_ACCEL_DEFAULT_RATE, ADIS16448_ACCEL_DEFAULT_DRIVER_FILTER_FREQ),
	_accel_filter_y(ADIS16448_ACCEL_DEFAULT_RATE, ADIS16448_ACCEL_DEFAULT_DRIVER_FILTER_FREQ),
	_accel_filter_z(ADIS16448_ACCEL_DEFAULT_RATE, ADIS16448_ACCEL_DEFAULT_DRIVER_FILTER_FREQ),
	_mag_filter_x(ADIS16448_MAG_DEFAULT_RATE, ADIS16448_MAG_DEFAULT_DRIVER_FILTER_FREQ),
	_mag_filter_y(ADIS16448_MAG_DEFAULT_RATE, ADIS16448_MAG_DEFAULT_DRIVER_FILTER_FREQ),
	_mag_filter_z(ADIS16448_MAG_DEFAULT_RATE, ADIS16448_MAG_DEFAULT_DRIVER_FILTER_FREQ),
	_accel_int(1000000 / ADIS16448_ACCEL_MAX_OUTPUT_RATE, true),
_gyro_int(1000000 / ADIS16448_GYRO_MAX_OUTPUT_RATE, true),
```

```
case SENSOR_POLLRATE_DEFAULT:
				return ioctl(filp, SENSORIOCSPOLLRATE, ADIS16448_ACCEL_DEFAULT_RATE);

			/* adjust to a legal polling interval in Hz */
			default: {
					/* do we need to start internal polling? */
					bool want_start = (_call_interval == 0);

					/* convert hz to hrt interval via microseconds */
					unsigned ticks = 1000000 / arg;

```

1.  Why is the sampling frequency so low？？

     unsigned ticks = 1000000 / arg; => ticks = 1000000 / 100 = 10000us = 10ms
 
2.  I don't think it's reasonable：
```
_accel_int(1000000 / ADIS16448_ACCEL_MAX_OUTPUT_RATE, true),  
bool accel_notify = _accel_int.put(arb.timestamp, aval, aval_integrated, arb.integral_dt);
```
It should be like this：
```
_accel_int(1000000 / (ADIS16448_ACCEL_DEFAULT_RATE, true),  
bool accel_notify = _accel_int.put(arb.timestamp, aval, aval_integrated, arb.integral_dt);
```

dagar, 2017-12-26:
The adis16448 driver was recently contributed by @ASM3.

ASM3, 2018-01-04:
The ADIS16448 has an internal filter and a averaging decimation filter which is set to 100 Hz (application specific). Thus, the IMU delivers data at 100Hz and the driver is set accordingly. If you need another sampling rate or filter configuration just set the driver according to your application.
The integral reset is set according to the maximal possible sampling frequency of the ADIS16448 which is ~819 SPS.

yangang123, 2018-01-08:
@ASM3. thank you for your reply.

dagar, 2018-01-08:
I think we can close this now.

closed 2018-01-08 by dagar