PX4/PX4-Autopilot #8527: [Bug] Brake twitching on huge multicopter( wheelbase >800mm) for new V1.7.2 code

whyscience, 2017-12-26:
Log is here: https://logs.px4.io/plot_app?log=e9e92a61-7db0-487c-99cc-ad68b171e0d7

[image]

[image]

Using the default params, the wheelbase is 800mm and 1200mm, when the drone brakes, it brakes twice or triple, as shown in the log.
On DJI F450, it's OK.

whyscience, 2017-12-26:
- Another log: https://logs.px4.io/plot_app?log=6d4f4e0d-7707-44b9-a673-057fbb2a4a15
- The video is here: https://youtu.be/tuYLPnBJfkk

LorenzMeier, 2017-12-26:
The controller goes into saturation on the outputs:
https://logs.px4.io/plot_app?log=6d4f4e0d-7707-44b9-a673-057fbb2a4a15#Nav-Actuator-Outputs-Main

This is not a Firmware issue, but a tuning issue on your airframe. We have improved the control performance in V1.7 and you might need to adjust your attitude / rate control gains.

closed 2017-12-26 by LorenzMeier

whyscience, 2017-12-26:
OK, thanks.

whyscience, 2017-12-26:
@LorenzMeier  I still have a doubt, there is only one saturation in the log, but in the video, it twitched more than one time, the corresponded log didn't go to saturation. 
[image]

LorenzMeier, 2017-12-26:
Your system is in [marginal stability](https://en.wikipedia.org/wiki/PID_controller#Stability) and you need to tune it. The parameter reference has some explanations, but in general familiarity with PID controllers is required.

I can see that you have not tuned the PID gains at all for your airframe and used the default gains. This *will not work* (it never works for any flight controller). If you fly a big drone you absolutely need to tune the control loops. The defaults only work for default-sized drones.

Tuning gain reference is here:
https://dev.px4.io/en/advanced/parameter_reference.html#multicopter-attitude-control

If you do not know how to tune you can drop me an email to lorenz@px4.io and I can put you in touch with people in the community that offer commercial support. You can alternatively post to http://discuss.px4.io for help from the community.

This is not a software issue but an issue with your airframe setup, so we will not continue to analyse it here. Thanks!

whyscience, 2017-12-26:
Thanks again.