PX4/PX4-Autopilot #8529: pixhawk can not startup when flash v1.7.2 firmware

jamming, 2017-12-27:
pixhawk can not boot up when flash v1.7.2 firmware. if remove R635 it can boot up.
R635 is a reset resistor.

LorenzMeier, 2017-12-27:
@jamming I've sent you an email regarding this requesting test hardware. @davids5 @dagar This is what we discussed on the dev call.

Mecher, 2017-12-28:
Same problem here , works well with APM

tops4u, 2017-12-28:
I have a similar Problem so I post it here, please let me know if it is related or otherwise I'll open a new Issue. Also my Issue may be HW or only Bootloader Code related. 

Booting with Battery works without any Issues, booting only with USB Power fails 90% of Times when connected to a MacBook, Avionics Rail will only get about 4.2 - 4.8V from the Macbook for a reason I don't know (the OS does not complain about over current on the USB Port). When using a 5V / 2A Power supply it will eventually Boot somtimes within 10 secs, after 2 Minutes or not at all - in this Case Avionics Power will show 5V. When booting with the Battery and Powering over an ACSP-4 PDB Avionics will get 5.3V. To me it looks as if the FMU is stuck in the Bootloader (IO blue is showing HB, red IO Led is steady on, while the FMU LED is flashing very fast and the Safety Button flashing fast). This is on a mRo X-2.1 Revision 2 Unit. Reverting to old Versions of PX4 do not show this pattern.

closed 2018-03-19 by thomasgubler