PX4/PX4-Autopilot #8530: micrortps_bridge: uorb multi-topics don't work

ChristophTobler, 2017-12-27:
Currently, uorb multi-topics (e.g. `optical_flow_raw`, `vehicle_vision_position`, etc.) don't work with the micrortps_bridge. 
https://github.com/PX4/Firmware/blob/master/src/modules/micrortps_bridge/CMakeLists.txt

related to https://github.com/PX4/Firmware/pull/8495. But I guess we need to fix it anyways.

@dagar I saw you've worked on it. Is that something you could look into?

dagar, 2017-12-28:
I'm not that familiar with FastRTPS itself, but at a glance it doesn't look like the concept of multi-topics exists. I think we'll need to treat them like individual topics (sensor_accel_0, sensor_accel_1, etc), but it could be done automatically. 

The manually maintained message ids (https://github.com/PX4/Firmware/blob/master/msg/tools/uorb_rtps_message_ids.py#L5) are already a problem. So the entire thing needs to be considered.

Should we discuss this on the messaging working group call?

ChristophTobler, 2017-12-29:
>Should we discuss this on the messaging working group call?

Sure. When's that?

dagar, 2017-12-29:
It's on Thursday, although I've never attended. 
https://dev.px4.io/en/contribute/#calendar

The big overall issue that's not at all clear to me is how we need to handle message versioning and ids when you start using them outside of Firmware.

TSC21, 2018-08-10:
@santiago3dr can we please keep this open? This is something that still needs to be tackled.

dagar, 2019-02-11:
@TSC21 I believe you'll need to explicitly handle the instances as separate topics.

TSC21, 2019-02-11:
> @TSC21 I believe you'll need to explicitly handle the instances as separate topics.

Yes I know. Just didn't have time to implement this yet.

TSC21, 2019-06-02:
Solution in https://github.com/PX4/Firmware/pull/12137

TSC21, 2019-08-31:
Solved already.

closed 2019-08-31 by TSC21