PX4/PX4-Autopilot #8531: Overshoot on setpoint of position

BazookaJoe1900, 2017-12-27:
its seems that the position setpoint work incorrectly. it has some overshoot at the end all the time.
For example:
[image]
taken from log:
https://logs.px4.io/plot_app?log=62842ca9-1d3c-4b5a-8057-8ff30ffbdb10

On my configuration those overshoots are much larger.

In general I think that the position setpoint should be 'steps' and not gradual that depend on the velocity parameters. Of course that there is internal slew rates, but the "position setpoints" are 'steps'

I think that there were some issues about that in the past, I just can't find them...

dagar, 2017-12-28:
@Stifael @MaEtUgR

Stifael, 2017-12-28:
@BazookaJoe1900 on which configuration?
Those "overshoots" are not really overshoots. The position setpoint that you see in auto consists of two parts: 1). the position of the closest point on the line previous-current-triplet setpoint. 2.) the cruise-speed of the vehicle. Usually the cruise-speed is used as feedforward velocity in the velocity controller. However, px4 did not support velocity feedforward properly (but soon it will), which led to the above implementation where cruisee-speed is inverted to position through the position-controller P-Gain [here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2222-L2223). In addition, px4 uses triplets for navigation, previous to current and next to current setpoints. The speed at the current setpoint is defined by the angle previous-current-next, where at 180, the vehicle will just pass the current setpoint without stopping. The speed at 90 degrees is controller by [MPC_CRUISE_90](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_params.c#L260), and all other angles are mapped from a quadratic function. Hence, at an angle of 0, the vehicle will target zero cruise speed at current setpoint, and at an angle of 180 it will target maximum cruise speed which looks like an overshoot in position...

Stifael, 2018-01-23:
@BazookaJoe1900 I will close it for now. You can reopen it if you think that it is necessary.

closed 2018-01-23 by Stifael